CadQuery/cadquery #244: Code formatting and lint checks

Peque, 2019-12-15:
Would you be willing to accept a PR implementing auto-code-formatting? I would suggest [Black](https://github.com/psf/black).

I could also add some lint checks (i.e.: `flake8`) and maybe `isort`. It would mean many lines of code changed, but I think it is worth it. :blush:

jmwright, 2019-12-15:
@Peque Will Black make changes to the user facing API (i.e. method names)? If so, would it be possible to disable those format checks for now? It's going to be a lot of work to make the transition to standard Python names for methods. If we can come up with a good plan for that transition it would help, but it will still be hard with the limited time for CadQuery that we all have.

Referencing another issue on formatting: https://github.com/CadQuery/cadquery/issues/195

Peque, 2019-12-15:
@jmwright Yeah, Black would not complain about names. `flake8` would not complain either by default.

I hope one day CadQuery adopts PEP8 for its API, but that is not my goal with this proposal. :blush:

jmwright, 2019-12-16:
@Peque Would the code formatting changes be made to code without any user interaction, or would they be another check in the CI pipeline? What would the workflow look like?

Peque, 2019-12-16:
@jmwright I would say just a check in the CI pipeline for now. User would need to run it manually or configure their code editor to automatically format their code for them.

jmwright, 2019-12-16:
Installed black to try it out and immediately hit this: https://github.com/psf/black/issues/1207

We'd have to pin the regex version for now in the CI environments to get it to run.

Peque, 2019-12-16:
@jmwright What a bad luck! :joy: 

It seems you hit that issue the moment it first occurred! (it was reported 2 hours ago). It seems it is fixed already, but I have not tried.

By the way, there are 2 parameters I consider "important" to set when using Black. One is the line length. It seems CadQuery is usually bellow 79, so I would set it to 79 (that is not the default). The other is the quotes. By default Black will use `"` (double). It seems CadQuery base code is not consistent here, so I would say the default is just fine. :blush:

jmwright, 2019-12-17:
It doesn't appear that they've fixed it in PyPi yet. I tried an uninstall and reinstall and it didn't work. I'll try again tomorrow.

jmwright, 2019-12-18:
I went ahead and manually rolled the regex version back as other people on the black issue were doing.

@adam-urbanczyk Below are samples of the kinds of things that were changed by running it on our codebase. Mostly fixing quote usage and line breaks.

[image]
[image]
[image]

@Peque We have multiple open PRs right now.

Peque, 2019-12-18:
@jmwright Unless you have something against Black, I would not try to postpone this until all open PRs are closed. That may never happen. :smile: 

And I personally would not care about having to rebase my PRs if this got integrated before.

adam-urbanczyk, 2019-12-20:
@jmwright looks fine, we could adopt black as far as I'm concerned.

@Peque why wouldn't you just use the default parameters? I assume that they were chosen for a reason. 

Last but not least, how do we want to enforce this? Pre-commit hook? CI check?

Peque, 2019-12-20:
@adam-urbanczyk The reason 88 was chosen is because the author wanted it that way (justification is rather weak in my opinion). :joy: That is also the reason why projects like [White](https://github.com/kennethreitz-archive/white) exist. And why many projects change Black's defaults (in particular the max-line-length and the quotes-style).

The reason why I, personally, would suggest 79, it is because current CadQuery's code base seems to be closer to that 79-limit style, so it will probably result in less lines of code changed after formatting. That would be an objective reason I think, although maybe I should validate that hypothesis (compare the diffs when using 79 and 88).

I think CI should be mandatory (it is the ultimate check before integration). :blush:

Peque, 2019-12-20:
So a quick tests on `black -l [79|88] cadquery/` and then `git diff | wc -l` shows:

- 79: 4342 lines changed
- 88: 4125 lines changed

It seems I was wrong. Things like this (79 limit):

```diff
-    def workplane(self, offset=0.0, invert=False, centerOption='CenterOfMass',
-                  origin=None):
+    def workplane(
+        self,
+        offset=0.0,
+        invert=False,
+        centerOption="CenterOfMass",
+        origin=None,
+    ):
```

Are formatted instead like (88):

```diff
-    def workplane(self, offset=0.0, invert=False, centerOption='CenterOfMass',
-                  origin=None):
+    def workplane(
+        self, offset=0.0, invert=False, centerOption="CenterOfMass", origin=None
+    ):
         """
```

I still prefer the 79 limit, but at this point it is 100% subjective, so you decide. :wink:

Peque, 2019-12-22:
Quick test with just Black formatting (left the default max. line length since it resulted in less code changed): https://github.com/CadQuery/cadquery/pull/255

helmecke, 2020-05-10:
This issue should be closed, as the related PR is merged into `master`.

adam-urbanczyk, 2020-05-10:
Indeed @helmecke. Solved by #255 , thanks again @Peque!

closed 2020-05-10 by adam-urbanczyk

Peque, 2020-05-10:
@helmecke @adam-urbanczyk Only code formatting was merged. This issue could be left open until link checks are added as well (i.e.: `flake8`, `isort`...). Or another issue could be created for that, I do not mind. :blush:

Unfortunately, I do not think I will be able to work on this in the short term, so someone else should take over it. :innocent: