CadQuery/cadquery #246: SVG export does not support user-defined orientation

adam-urbanczyk, 2019-12-17:


jmwright, 2019-12-17:
Reference discussion here: https://groups.google.com/d/msg/cadquery/H_8RvlJ__bc/R5IBK6ATAgAJ

Right now the projection direction vector is hard coded, and it would be nice for users if they could adjust it. It would also be helpful if they could turn the origin marker on and off (would help with post processing such as laser cutter gcode generation), and if the origin marker matched the projection direction. It is currently also hard coded directly into the SVG template and is always in the same orientation.

Shimwell, 2020-07-16:
This is a nice feature request, I've been reading up on the google groups thread and, I just tried to change the DEFAULT_DIR setting and this certainly did change the camera angle, I was also able to remove the x,y,z axis which was also discussed in the google groups.

Once the dir was changed I noticed that the lines also look a little strange even with the new viewing angle.

I see from the SVG template that there are [solid lines](https://github.com/CadQuery/cadquery/blob/10118e563add4e60f21c882ff2a842757ceaacad/cadquery/occ_impl/exporters.py#L452) and [hidden lines](https://github.com/CadQuery/cadquery/blob/10118e563add4e60f21c882ff2a842757ceaacad/cadquery/occ_impl/exporters.py#L447). 

It appears that the solid lines and hidden lines are selected for the original default_dir

Does anyone have any tips to change which lines are solid and which are hidden when the default_dir also changes

Thanks

Shimwell, 2020-07-16:
I also tinkered around with the height and width settings which might be another nice property to expose to the user

jmwright, 2020-07-16:
CQ's SVG export is still pretty lacking, and there's no control of solid vs hidden that I know of in the CQ API. However, now that we have more direct access to the OCCT kernel it opens up some new possibilities. If the required interfaces are available in OCP it should be possible to drop down to a lower level and do custom SVG export.

https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_visual3d___view.html

adam-urbanczyk, 2020-10-14:
Great, looking forward @ArmoredBlood

jmwright, 2021-02-05:
This was addressed in #596

closed 2021-02-05 by jmwright