CadQuery/cadquery #250: Test discovery does not work out of the box

adam-urbanczyk, 2019-12-20:
When running `pytest` I get

```
===================================================================================== test session starts ===================================================================================== 
platform win32 -- Python 3.6.8, pytest-5.3.2, py-1.8.0, pluggy-0.12.0
rootdir: C:\Users\lo-re\cq\cadquery, inifile: pytest.ini
collected 0 items

==================================================================================== no tests ran in 0.16s ====================================================================================
```

The solution is to add a `pytest.ini` with the following content

```
[pytest]
python_files = tests/Test*.py
```

adam-urbanczyk, 2019-12-20:
@Peque any clue why is this happening?

adam-urbanczyk, 2019-12-21:
I cannot reproduce it anymore in a new conda env - extremely weird.

Peque, 2019-12-21:
@adam-urbanczyk Maybe you had an outdated revision? Since you needed to:

```
[pytest]
python_files = tests/Test*.py
```

Note the `Test*.py` which corresponds with the old file names. New tests are named `test_*.py`, so `pytest` should be able to discover them automatically. :blush:

adam-urbanczyk, 2019-12-21:
Yep, looks like it. Sorry for the noise.

closed 2019-12-21 by adam-urbanczyk