CadQuery/cadquery #254: Rotation axis changed?

bragostin, 2019-12-21:
With Cadquery master from Sept 2019, using ```rotate=cq.Vector(0, 45, 0)``` closed the wire below:
```python
edge_points = [[-7.,-7.,0.], [-3.,-10.,3.], [7.,-7.,0.], [7.,7.,0.], [-7.,7.,0.]]
edge_wire = cq.Workplane('XY').polyline([(-7.,-7.), (7.,-7.), (7.,7.), (-7.,7.)])
edge_wire.add(cq.Workplane('YZ').workplane().transformed(offset=cq.Vector(0, 0, -7), rotate=cq.Vector(0, 45, 0)).spline([(-7.,0.), (3,-3), (7.,0.)])) 
show_object(edge_wire)
```
In CadQuery master from Dec 2019 only ```rotate=cq.Vector(45, 0, 0)``` closes the wire. Have rotation axis changed lately?

jmwright, 2019-12-21:
@bragostin I'm guessing this issue is related: https://github.com/CadQuery/cadquery/pull/243#discussion_r360658424

bragostin, 2019-12-21:
@jmwright I think the previous logic was correct, I can't find a reasonning to use ```rotate=cq.Vector(45, 0, 0)``` in this example.

jmwright, 2019-12-21:
@bragostin Would you mind joining the conversation on that issue?

Peque, 2019-12-22:
@bragostin What is the expected output of that code? For what I understand, there should be no rotation? :thinking: 

```python
import cadquery as cq
from jupyter_cadquery.cadquery import show

edge_wire = cq.Workplane('XY').polyline([(-7.,-7.), (7.,-7.), (7.,7.), (-7.,7.)])
edge_wire.add(cq.Workplane('YZ').workplane().transformed(offset=cq.Vector(0, 0, -7), rotate=cq.Vector(0, 0, 0)).spline([(-7.,0.), (3,-3), (7.,0.)])) 
show(edge_wire)
```

[image]

bragostin, 2019-12-22:
This is the expected result with ```rotate=cq.Vector(0, 45, 0)```, and what I obtained with CadQuery master from Sept 2019:
[image]
But with CadQuery from 22 Dec 2019 I get:
[image]
To obtain the expected result I must now use  ```rotate=cq.Vector(45, 0, 0)```
It seems that the rotation axis are kind of messed up now, because it seems logical that ```y``` should be the rotation axis to obtain the expected result.

Peque, 2019-12-22:
@bragostin Thanks for sharing the expected output. I understand now.

That was a broken behavior and was (only partially) fixed with https://github.com/CadQuery/cadquery/pull/235. Should be completely fixed with https://github.com/CadQuery/cadquery/pull/243.

Basically, rotations are relative to the axis of the plane you are rotating. So yeah, you should rotate over X, since the plane you created is the YZ plane, not the XY plane.

bragostin, 2019-12-22:
@Peque thank you for the clarification! My intention was to tilt the 'YZ' plane around the Y axis. Shouldn't it be possible to rotate a plane with any axis?

Peque, 2019-12-22:
@bragostin The moment you create an YZ plane, you have a different coordinates system. See for example your `.spline([(-7.,0.), (3,-3), (7.,0.)]))`: those are your `(x, y)` points in the new coordinates system. You want to rotate around X, which is the axis passing through `(-7, 0)` and `(7, 0)`.

Is there any use case you cannot solve after the latest changes? Or a use case that is significantly harder to solve? Note that those changes did solve the problem of being unable to stack/concatenate multiple rotations in an easy way.

bragostin, 2019-12-22:
@Peque with the current master branch I still observe the same behaviour. 
Do you mean that the current behaviour is going to be the new normal? This may create backward compatibility issues. If I have to changes all the rotation axis in my models to the new norm that's quite a lot of work.

Peque, 2019-12-22:
@bragostin What do you mean by "the same behavior"? If you mean that, for your example provided above, you need to rotate around X and not Y, then that is correct.

Unfortunately this is a backwards-incompatible change, yeah. The old behavior was broken, unfortunately. :sweat:

bragostin, 2019-12-22:
@Peque I am afraid I still don't understand the new logic here. With CadQuery from Sept 2019 it was possible to rotate a plane using an axis of the global coordinate system, if this possibility disappears, I am afraid that more unexpected effects will appear. Is it not possible to still allow rotation with an axis of the global coordinate system as an option maybe?

Peque, 2019-12-22:
@bragostin I do not know CadQuery's API deeply, so I cannot answer that question. @adam-urbanczyk may though. :blush:

Note that, before, you were not really rotating using an axis of the global coordinate system. Your plane was transformed with an offset, so you were already using a different (local) axis, which happened to be parallel to global-Y. The difference is that what you called Y is now X (as it should be in the local coordinates system).

Note that you can always redefine your problem to be able to rotate around local Y, although then you need to change even more code:

```python
import cadquery as cq
from jupyter_cadquery.cadquery import show

edge_wire = cq.Workplane('XY').polyline([(-7.,-7.), (7.,-7.), (7.,7.), (-7.,7.)])
edge_wire.add(cq.Workplane('XY').workplane().transformed(offset=cq.Vector(-7, 0, 0), rotate=cq.Vector(0, -45, 0)).spline([(0.,-7.), (-3,3), (0.,7.)])) 
show(edge_wire)
```

bragostin, 2019-12-22:
@Peque OK I get it thank you for taking the time to explain! 
Now I just imagine others users like me who have previously working models that will now stop working and will have no clue why.
Isn't it possible to provide some warning information explain the issue when using rotate?

Peque, 2019-12-22:
@bragostin It should definitely be noted in the CHANGELOG or the release notes in the next release. Not sure a warning should be added in-code though, specially considering CadQuery is still RC.

In my opinion fixing bugs should allow for backwards-incompatible changes without "deprecation warnings". But it is not up to me to decide it. :innocent: 

@adam-urbanczyk @jmwright

jmwright, 2019-12-22:
@adam-urbanczyk and @dcowden have a much deeper understanding of things like this than I do.

From a users perspective, what if there was something like a `legacy` argument that provided the old behavior? It would be set to `False` by default so that the behavior is not backwards compatible. This would provide a relatively easy fix for users who are expecting the old behavior. A deprecation warming could be set up for that argument, letting users know that it will be removed in a future release.

Thoughts?

Peque, 2019-12-22:
Just as a joke: https://xkcd.com/1172/ :joy: 

I would say that adding that parameter is not a good idea. It is harder to implement and support and it only delays the inevitable: the user needs to fix their code. Old behavior was broken and there were no tests for those plane rotations; I would not encourage users to keep their legacy code.

Instead of being lazy and use `legacy=False`, just fix your code. Or use an older version, where you will always find that broken behavior, if that is what you want (knowing that you will get no support though). :stuck_out_tongue_winking_eye: 

Just my opinion, so whatever you say/decide is fine with me. :blush:

bragostin, 2019-12-23:
I guess fundamentally Peque is right :)
I have confused a bug with a feature: the possibility to rotate planes around what I thought was a global axis.
If I am the only one who did this mistake then it does not matter, there is nothing else to do. But if other users are in the same case, I think they should receive some information to understand why their previously working models are now crashing, otherwise they will come here and start publishing issues ;)

jmwright, 2019-12-23:
A short description of the potential gotcha could be added to the docstring so that the information is embedded in the docs. Then we can just link to that if it comes up.

adam-urbanczyk, 2019-12-23:
I think we were a little bit too fast with this issue. I'd propose the following way out:

1. Fix the underlying code for rotation and enable both relative and absolute rotations
2.  Add a `relative` parameter to `transformed` and set it to `False` by default with a warning that it will be changed in CQ2.1
3. In the (not yet finalized) CQ2.0 release still use the `relative=False`
3. Update the documentation accordingly

Any thoughts?

Peque, 2019-12-23:
@adam-urbanczyk Note that:

- Both rotations, the old and the new one, are relative (i.e.: the old rotations were not around global axis either), so maybe `relative` is not the best name (?)
- New behavior matches the old (and current) docstring description (i.e.: old behavior was not just different from the new, but broken according to specifications in the documentation)
- There are no tests for the old behavior, so maybe some should be created if supporting that feature
- I think I would not enjoy implementing an extra parameter to support the old behavior, so I will not volunteer to make the change and revert it for CQ2.1, hope you understand :innocent: :heart: Feel free to merge my PR first, or push changes to it (you should have permission to push to that branch in my fork) or simply close it if you would rather start from 0. I think the tests can be reused though.

Thinking about possible API changes (just what comes to my mind as I write this :joy: ). Since `.rotate((x, y, z))` is simply a concatenation of multiple rotations, maybe rotations could be constricted to one axis only. This simplifies the API a bit, knowing that supporting both global and relative rotations would already make things more complicated:

```python
def rotate(
    axis: Union[str, Vector],
    angle: float,
    origin: Tuple = (0, 0, 0),
    global_axis: bool = False,
    global_origin: bool = False
):
    pass

.rotate('x', 42)  # Relative  42 degrees around local X
.rotate((1, 0, 0), 42)  # Rotates 42 degrees around local X
.rotate('y', 42, global_axis=True)  # This would be like the old behavior, with only relative origin
.rotate('z', 42, global_origin=True)  # Rotates over global (0, 0, 0), local Z
...
```

But already looks like too complicated. :sweat_smile: 

Maybe it would be easier if we knew an example that needs absolute rotations (i.e.: with respect to global coordinates). Can't they be performed by starting with a rotated plane around the global origin instead? The examples we have for now all can be solved with relative rotations.

bragostin, 2019-12-23:
In order to minimize the amount of extra work, how about just having a temporary warning in the ```transformed``` function explaining the bug correction? The warning would then be disabled in CQ2.1.
In my case I work with both the September and December versions of CadQuery now, so that could be a potential mitigation for other users too.

bragostin, 2020-01-20:
With the code below I obtain a different end results for part_OK and part_NOK as shown on the picture below:
```python
import cadquery as cq
part = cq.Workplane('XY').box(140, 64, 8)
part_NOK = part.faces('>Z').workplane(offset=0).transformed(rotate=(0, 45, 0)).rect(10, 20).extrude(30)
part_OK = part.faces('>Z').transformed(rotate=(0 ,45 ,0)).rect(10, 20).extrude(30)
show_object(part_NOK)
show_object(part_OK)
```
Is this to be expected?
[image]

Peque, 2020-01-20:
@bragostin I see some slight unexpected rotation, right? I think that should be fixed with https://github.com/CadQuery/cadquery/pull/243.

@adam-urbanczyk I think https://github.com/CadQuery/cadquery/pull/243 could be merged now, while the new API is discussed/implemented.

bragostin, 2020-01-20:
@Peque That's right, the two inclined boxes should be coincident I would think.

Peque, 2020-01-20:
@bragostin If you try #243 you will see the rotation fixed.

Boxes will not be coincident though. One will be translated `4` from the other in the Z direction. That may be expected (i.e.: if you do not call `.workplane()` it seems the workplane is the X-Y plane were you created the `part` box, one you call `.workplane()` you make sure to work from a plane at the `>Z` face).

bragostin, 2020-01-20:
@Peque thank you!
Yes I already noticed that sometimes if you call ''faces(selector)'' without ''.workplane()" then I am still on the original plane. I wonder if this is as it should be?

Peque, 2020-01-20:
:woman_shrugging:

jmwright, 2020-01-21:
The logic on whether to call `.workplane()` was before my time, and I've always just accepted it. When I create tutorials and examples, I try to remember to add a workplane call in the chain right after a selector. That seems to be the safest way to ensure that you always get the expected behavior.