CadQuery/cadquery #257: Hello, can we update the docker hub images or create a new tag with python37?

Dustyposa, 2019-12-27:
My env has something wrong.
I run the below code in my local computer and in docker images, but the output is different:
```python
import json
import cadquery as cq
import io

def export_edgedata_to_json(edge_hash, point_set):
    points_coordinates = []
    for point in point_set:
        for coord in point:
            points_coordinates.append(coord)
    edges_data = {"metadata": {"version": 4.4,
                               "type": "BufferGeometry",
                               "generator": "ding_dang_maker"},
                  "uuid": edge_hash,
                  "type": "BufferGeometry",
                  "data": {"attributes": {"position": {"itemSize": 3,
                                                       "type": "Float32Array",
                                                       "array": points_coordinates}
                                          }
                           }
                  }
    return json.dumps(edges_data)

def print_face_or_line_data(str_to_out):
    model_type = 1
    res = {'type': model_type, 'data': str_to_out}
    print(json.dumps(res))

def show_model(model, exporter, tolerance=0.1):
    s = io.StringIO()
    model_type = 0
    exporter.exporters.exportShape(model, exporter.exporters.ExportTypes.STL, s, tolerance)
    res = {'type': model_type, 'data': s.getvalue()}
    with open('rssssu.stl', 'w') as fp:
        print(json.dumps(res), file=fp)

size = 10

def make_rubish(x, y, z):
    rubish = cq.Workplane("XY").workplane(offset=z * size).center(x * size, y * size).rect(10 * size,
                                                                                           10 * size).extrude(15 * size)
    rubish = rubish.faces(">Z").shell(0.5 * size)

    box1 = cq.Workplane("XY").workplane(offset=(9 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,
                                                                                              1 * size)
    box2 = cq.Workplane("XY").workplane(offset=(12 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,
                                                                                               1 * size)
    rubish = rubish.cut(box1).cut(box2)

    show_model(rubish, cq)

make_rubish(0, 0, 0)
```

Dustyposa, 2019-12-27:
In docker images, i get a 3d model:
[image]
In local computer stl model like this:
[image]

So, I need to get a new env that can check what happen, to certain that should me to update the docker images.
Then I `docker pull conda/miniconda3` it's fast.  
But when i run `conda install -c conda-forge -c cadquery cadquery=2`, it's to slow.
Can we update the cadquery docker images that i can pull a new version fastly?  :smile:
Thanks in advance.

jmwright, 2019-12-28:
There is a 3.7 dockerfile on the fork [here](https://github.com/roipoussiere/cadquery/blob/master/Dockerfile.python3.7), but I haven't tested it. @roipoussiere do you have any guidance on this?

Dustyposa, 2019-12-28:
I tried it in docker container.But the model didn't change.
So, the current stl shape is that in latest cadquery?
First one or second one?
Now I'm trying to build a docker images with latest python-occ and cadquery to check why stl shape is different.
And the system (linux/windows) has any effect?

Dustyposa, 2019-12-28:
Emmm, I'm failed.
The latest `python-occ 7.4`'s some api changed. `cadquery` can't work with it.
Have any idea?
The same code get two different model shape?

jmwright, 2019-12-28:
@Dustyposa This is what I get when I run your script from within CQ-editor.
[image]
The STL that your script creates won't open in FreeCAD or PrusaSlicer for me. I modified the script and now I get an STL that works:
```python
import json
import cadquery as cq
import io

def show_model(model, exporter, tolerance=0.1):
    s = io.StringIO()
    model_type = 0
    exporter.exporters.exportShape(model, exporter.exporters.ExportTypes.STL, s, tolerance)

    with open('rssssu.stl', 'w') as fp:
        fp.write(s.getvalue())

size = 10

def make_rubish(x, y, z):
    rubish = cq.Workplane("XY").workplane(offset=z * size).center(x * size, y * size).rect(10 * size,
                                                                                           10 * size).extrude(15 * size)
    rubish = rubish.faces(">Z").shell(0.5 * size)

    box1 = cq.Workplane("XY").workplane(offset=(9 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,
                                                                                              1 * size)
    box2 = cq.Workplane("XY").workplane(offset=(12 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,
                                                                                               1 * size)
    rubish = rubish.cut(box1).cut(box2)

    rubish.val().exportStl('/home/jwright/Downloads/rssssu.stl')

    show_object(rubish)

    # show_model(rubish, cq)

make_rubish(0, 0, 0)
```
And here's the STL after it's opened in FreeCAD. There's nothing special about using FreeCAD, I was just testing the STL.
[image]
The exporters interface needs to be improved, and there's an issue on it [here](https://github.com/CadQuery/cadquery/issues/216).

Dustyposa, 2019-12-28:
@jmwright Thanks for your reply. 
And the code should get first one is right or wrong? (model1)
Because I cut box twice and the  box used to cut should disappear right？

jmwright, 2019-12-28:
@Dustyposa I see what you're saying now. Yes, it's intuitive that you could chain the two `cut` calls and have two cutouts. I'll have to investigate what's going on here. In the meantime, something like the following should work, and is normally how I would do it (favoring `cutBlind` and `cutThru` operations over boolean operations like `cut`.
```python
import json
import cadquery as cq
import io

size = 10

def make_rubish(x, y, z):
    rubish = cq.Workplane("XY").workplane(offset=z * size).center(x * size, y * size).rect(10 * size,
                                                                                           10 * size).extrude(15 * size)

    rubish = rubish.faces(">Z").shell(0.5 * size)

    rubish = (rubish.faces("<Y").workplane(centerOption="ProjectedOrigin").transformed(rotate=(0, 90, 0))
                               .center(-35, 0).rect(1 * size, 7 * size)
                               .center(-70, 0).rect(1 * size, 7 * size)
                               .cutThruAll())

    #rubish.val().exportStl('/home/jwright/Downloads/rssssu.stl')

    return rubish

rubish = make_rubish(0, 0, 0)

show_object(rubish)
```

adam-urbanczyk, 2019-12-29:
If I use CQ-editor 0.1RC1 and the following code (I removed all exporting to disk stuff):
```python
size = 10

def make_rubish(x, y, z):
    rubish = cq.Workplane("XY").workplane(offset=z * size).center(x * size, y * size).rect(10 * size,10 * size).extrude(15 * size)
    rubish = rubish.faces(">Z").shell(0.5 * size)
    box1 = cq.Workplane("XY").workplane(offset=(9 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,1 * size)
    box2 = cq.Workplane("XY").workplane(offset=(12 + z) * size).center(x * size, y * size).box(7 * size, 18 * size,1 * size)
    rubish = rubish.cut(box1).cut(box2)
    show_object(rubish)

make_rubish(0, 0, 0)
```
I get the expected result:

[image]

Dustyposa, 2019-12-30:
@adam-urbanczyk The cadquery is latest version? I use `start_distplay()` form occ can get the same result to on my local computer.
I will check my local env's cadquery source code to find the reason why that I can't get the correct result in my docker env.

adam-urbanczyk, 2019-12-30:
Nope, I used the packaged CQ-editor from https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1 which likely means CQ2RC1

Dustyposa, 2019-12-30:
Got it. Old version should be ok. Latest version has some wrong I guess.

adam-urbanczyk, 2019-12-30:
I can reproduce the issue in my local environment. It is not influenced by the version of CQ (HEAD or 2.0RC1) so I have to conclude that is has something to do with the build of oce and/or pythonocc.

lorenzncode, 2022-09-23:
I cannot reproduce this with current master or standalone CQ-editor 0.2.

adam-urbanczyk, 2022-09-23:
I cannot reproduce this anymore, closing.

closed 2022-09-23 by adam-urbanczyk