CadQuery/cadquery #259: `revolve` entire 360°

asprionj, 2019-12-28:
(I'm on 2.0RC1.) I want to create a simple torus, i.e. revolving a circle. If I revolve for anything shy of 360°, it works:
```
cq.Workplane('XY').circle(10).revolve(350, [-30,0,0], [-30,1,0])
```
[image]

However, if I want the full 360°, I somehow do not get a solid (?), at least it is shown very differently:
```
cq.Workplane('XY').circle(10).revolve(360, [-30,0,0], [-30,1,0])
```
[image]

What happened here? What's the difference? Is it intended?

adam-urbanczyk, 2019-12-28:
Confirmed, it is a bug. Funny enough the following works as expected:
```python
res = cq.Workplane('XY').rect(2,2).revolve(360, (-30,0,0), (-30,1,0))
```

adam-urbanczyk, 2019-12-28:
I found the following workaround:
```python
res = cq.Workplane('XY').circle(10).revolve(360, [-30,0,0], [-30,1,0],clean=False)
```

[image]

RubenRubens, 2020-02-05:
The bug is at exporting the geometry as STL. When it's exported as STEP it works fine.

**STL**
```python
import cadquery as cq
res = cq.Workplane('XY').circle(10).revolve(360, [-30,0], [-30,1], clean=True)
res.val().exportStl('test.stl', precision=0.1)
```

**STEP**
```python
import cadquery as cq
res = cq.Workplane('XY').circle(10).revolve(360, [-30,0], [-30,1], clean=True)
res.val().exportStep('test.step')
```
[image]

adam-urbanczyk, 2020-02-05:
@RubenRubens  I don't that this is the case:
```python
res = cq.Workplane('XY').circle(10).revolve(360, [-30,0], [-30,1], clean=True).val().isValid()
```
results in a `False`. It has nothing to do with the export. It is rather interesting that the STEP exporter does not mind.

RubenRubens, 2020-02-05:
Good to know. 👍

Shimwell, 2020-04-16:
I also stumbled across this bug today

A list of points connecting with a spline and rotated to 360 degrees doesn't export anything to the STL file but the STEP export worked fine.

The same list of points connected with a polyline exported fine to both STL and STEP. 

The same list of points connected with a mixture of spines and straight lines exported fine to both STL and STEP.

jmwright, 2020-04-16:
Just tested the code posted by @Shimwell on #322 with the OCP branch (and OCCT 7.4) and the stl export worked fine. I'm hoping to get some time to put a CQ 2.0 release together within the next few days. Once that has been published, we can merge the OCP branch and move to OCCT 7.4 for the CQ 2.1 release cycle. Numerous issues will be solved by moving to OCCT 7.4 in addition to this one.

[image]

adam-urbanczyk, 2020-06-04:
OCP is merged into master and the full rotation works as expected:
[image]

closed 2020-06-04 by adam-urbanczyk