CadQuery/cadquery #260: cut with solid from parametricCurve

asprionj, 2019-12-28:
(I'm on 2.0RC1.) Trying to cut a body that is constructed by revolving a `parametricCurve` from another body. For illustration, a minimal example which is working as expected, but using a `circle` instead of said `parametricCurve`:

```python
s = cq.Workplane('XY').circle(50).extrude(20) # the base body
to_cut = cq.Workplane('XY').transformed(offset=[35,0,0], rotate=[90,0,0]).circle(14).revolve(315, [-35,0,0], [-35,1,0]) # this will be cut away
s = s.cut(to_cut) # cut it...
```

The result looks fine (I cannot fully close the torus because of #259):
[image]

Now, if I use a `parametricCurve`, e.g. to have an ellipse instead of a circle (or any other complex geometry):

```python
from math import sin, cos, pi

def my_ellipse(t, a, b):
    return (a*sin(2*pi*t), b*cos(2*pi*t))

to_cut = cq.Workplane('XY').transformed(offset=[35,0,0], rotate=[90,0,0]).parametricCurve(lambda t: my_ellipse(t,14,7)).revolve(315, [-35,0,0], [-35,1,0])
``` 
it fails with "**RuntimeError**: Standard_Failure Courbes non jointives". (Also when using `intersect` instead of `cut`.)

Funny thing is, if I use `union` instead of `cut`, it does work as expected:
[image]

adam-urbanczyk, 2019-12-29:
Confirmed, looks like a cad kernel issue. Same workaround as in #259 :
```python
from math import sin, cos, pi

def my_ellipse(t, a, b):
    return (a*sin(2*pi*t), b*cos(2*pi*t))

to_cut = cq.Workplane('XY').transformed(offset=[35,0,0], rotate=[90,0,0]).parametricCurve(lambda t: my_ellipse(t,14,7)).revolve(360, [-35,0,0], [-35,1,0],clean=False)

s = cq.Workplane('XY').circle(50).extrude(20)
s = s.cut(to_cut,clean=False)

show_object(s)
```
[image]

lorenzncode, 2022-09-23:
Fixed in occt7.6 branch.

```py
from math import sin, cos, pi

def my_ellipse(t, a, b):
    return (a*sin(2*pi*t), b*cos(2*pi*t))

s = cq.Workplane('XY').circle(50).extrude(20) # the base body
to_cut = cq.Workplane('XY').transformed(offset=[35,0,0], rotate=[90,0,0]).parametricCurve(lambda t: my_ellipse(t,14,7)).revolve(315, [-35,0,0], [-35,1,0])
s = s.cut(to_cut)

print(s.val().isValid())
```
occt7.6:
[image]

current master (ocp-7.5.3):
[image]

adam-urbanczyk, 2022-09-23:
Thanks @lorenzncode for checking - I'm adding this issue to the PR.

closed 2022-10-01 by adam-urbanczyk