CadQuery/cadquery #262: 2019 CadQuery Community Update

jmwright, 2019-12-31:
# 2019 CadQuery Community Update

If you got a notification because you were mentioned in this issue and are not sure why, scroll down to the "Thank You" section. You are probably mentioned there. I apologize if anyone feels spammed, but the purpose is to say thank you for your contributions this project.

## Intro

2019 is quickly coming to an end, and it has been another big year for the CadQuery community. This year we moved to a new GitHub repository, and in the process changed the codebase over to rely on the PythonOCC API instead of FreeCAD. The majority of the port was done by @adam-urbanczyk as he has continued to lead our community in this new CadQuery 2.0 era. CQ 2.0 is still in the release candidate phase, but has already seen broad adoption by the community. A big thanks goes out to Adam for all of the time and effort that he has poured into the codebase and the community.

There are some issues that we have experienced with the current version of PythonOCC/OCE that we have been running on. @adam-urbanczyk has also been leading the charge on getting us updated to OCCT 7.3+ to address some of those issues and add new functionality. I'm hopeful that in 2020 we will see some major improvement on this front.

A note about CadQuery 1.0. I know that some users have built libraries and products on top of the FreeCAD version of CadQuery. The older FreeCAD-based version is still available, but the core CadQuery developers are very busy still with the CadQuery 2.0 transition, and will be for some time. If there are features that would help you transition over to CQ 2.0, please let us know in the [issue tracker](https://github.com/CadQuery/cadquery/issues). I would love to get all users transitioned to CQ 2.0 so that they can take advantage of the new features and bugfixes. This would also help keep the development team from being spread too thin.

## FOSDEM 2020

We will have a talk at [FOSDEM 2020](https://fosdem.org/2020/) in the [Open Source Computer Aided Modeling and Design](https://fosdem.org/2020/schedule/track/open_source_computer_aided_modeling_and_design/) devroom. If you have an interesting use for CadQuery that could be shared during the talk, please post a reply to [this issue](https://github.com/CadQuery/cadquery/issues/242). @adam-urbanczyk will be giving the talk, and it is entitled [Towards CadQuery 2.0](https://fosdem.org/2020/schedule/event/cadquery/). If you are near Brussels on Feburary 1st and 2nd, there is going to be a lot to do and see at FOSDEM, including a CadQuery talk! 

## Projects

Below are updates for many of the projects within the CadQuery community, including a couple that you might not have seen yet.

* [cadquery](https://github.com/CadQuery/cadquery) - The core library we all know. With PythonOCC compatibility has come an adjustment period, along with new features and development activity. 2020 should see a stable release and the feature set will continue to grow.
* [CQ-editor](https://github.com/CadQuery/CQ-editor) - This is the primary IDE for developing CadQuery scripts. It has support for advanced features like debugging and object inspection. MacOS support has been a challenge (see the call for help section below), but we definitely want MacOS support to be stable.
* [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) - If you are a Jupyter user or want to use CadQuery in the Cloud, this is a project you should check out. There is a version that runs on [MyBinder.org](https://camo.githubusercontent.com/483bae47a175c24dfbfc57390edd8b6982ac5fb3/68747470733a2f2f6d7962696e6465722e6f72672f62616467655f6c6f676f2e737667) which does not require a local installation of CadQuery. If you are just trying CadQuery out and do not want to set up Anaconda, this is a great option. I've started using jupyter-cadquery to create [online tutorials](https://github.com/CadQuery/cadquery-contrib/tree/master/tutorials).
* [cqparts](https://cqparts.github.io/cqparts/doc/) - cqparts is the creation of @fragmuffin and and is a wrapper around cadquery that allows for more complex parametric models to be created, including parts and assemblies. cqparts is not currently compatible with CadQuery 2.0, but there is an [issue](https://github.com/cqparts/cqparts/issues/144) related to the conversion.
* [fx-bricks fx-cad-notes](https://github.com/fx-bricks/fx-cad-notes) @michaelgale of @fx-bricks has put together a great resource for the community describing how they use CadQuery in their product development pipeline. Big thanks to @fx-bricks for publishing that for the community's benefit.
* [pywrap](https://github.com/CadQuery/pywrap) - This is a binding generator that @adam-urbanczyk has been working on, and the purpose is to auto-generate bindings for the OCCT CAD kernel, version 7.3 and newer. One of the challenges with many existing binding generation frameworks is that they require manual edits to get the bindings ready to use. The goal of pywrap is to have the OCCT bindings not require a single manual edit.

## Calls for Help

Here are two items of housekeeping for the community.

* @fragmuffin does not have much time to work on cqparts right now, and I'm sure that he would appreciate the help if any devs wanted to help him update the codebase for CadQuery 2.0. If you are interested in contributing to cqparts, please open an issue on the [cqparts repo](https://github.com/cqparts/cqparts/issues) and let @fragmuffin know.
* MacOS dev help needed: @adam-urbanczyk does not have access to a Mac, and I only have limited access to one. That has made it hard to troubleshoot issues with the cadquery library and CQ-editor on MacOS. It is also why there is not a CQ-editor release build in that repository. If you have a Mac and would like to do some deep troubleshooting, please let us know. If you are willing to take on handling the MacOS release builds of CQ-editor, please open an issue to let us know.

## Thank You

As always, a big Thank You goes out to all our contributors and users. There is an increasing amount of PRs being filed, and the amount of discussion within the community has been increasing as well. It's exciting to watch the community grow, and to see the creative ways people use CadQuery.

I also want to thank all the community members who answer questions for other users. The core devs are always happy to help, but it's a great sign of the health of a community when many members are helping each other and engaging in great discussions. I'm very proud of our community and how we have been working together.

I'm going to make an attempt at listing all the CadQuery community contributors here, but with many contributions spread across several repos in multiple projects, I run the risk of missing someone. I apologize if that happens. Please respond to this issue and let me know if I've missed a contributor or a project. I also do not have a good way to track participation in discussions currently, or community members who participate that way would be listed as well.

### CadQuery

@adam-urbanczyk
@hyOzd
@bragostin
@justbuchanan
@mgreminger
@huskier
@fragmuffin
@Peque
@bweissinger
@moeb
@westurner
@osterwood
@asukiaaa
@HLevering
@xix-xeaon
@phillipthelen
@krasin
@gebner
@bsilvereagle
@armyofevilrobots
@Grawp
@dcowden
@jmwright

### CQ-editor

@adam-urbanczyk
@justbuchanan
@gebner
@jmwright

### jupyter-cadquery

@bernhard-42
@ilya-epifanov

### cqparts

@fragmuffin
@zignig
@gntech
@justbuchanan
@luzpaz

michaelgale, 2019-12-31:
@jmwright Awesome wrap-up and many thanks to you and the rest of the core developers!  As an end user, I am truly grateful for the work you folks put into this project.  I have also noticed the increase in community activity over the past 2 years from the sidelines and share your optimism for this project's future because of its vibrant community.  

If you're looking for some assistance on the macOS side--I would be happy to offer what I can.  I have Apple dev credentials (good for making signed builds), reasonably comfortable with both XCode and *nix make tools, C/C++ etc., plus have a server full of VMs for cross validation and autobuilds.  I'd likely need learning curve to get up to speed with your dev/build pipelines, but hopefully I will become more useful with time!  Send me a PR or email via Fx Bricks.

adam-urbanczyk, 2020-01-01:
Thanks for such a nice summary @jmwright ! Thanks to our community 2019 was a good year for CQ and I sincerely hope that 2020 will be even better!

closed 2020-02-11 by jmwright