CadQuery/cadquery #263: RuntimeError: Standard_ConstructionError

greyltc, 2020-01-02:
```
import cadquery as cq

cube = cq.Workplane("XY").box(10, 10, 3)
cyl = cq.Workplane("XY").circle(5).extrude(3).translate((5, 0, -1.5))
u = cube.union(cyl)
```

Crashes with `RuntimeError: Standard_ConstructionError` 
This code at least will run to completion with `u = cube.union(cyl, clean=False)`

What's going on here?

jmwright, 2020-01-02:
@greyltc If you build `cube` and `cyl` in the same stack/chain instead of having them on separate Workplanes, does the `union` work?

greyltc, 2020-01-03:
@jmwright, I think this is what you mean:
```
import cadquery as cq

wp = cq.Workplane("XY")
stack = wp.circle(5).extrude(3)
stack = stack.box(10, 10, 3, centered=(True, False, True), combine=True, clean=True)
```
Yeah, that works (although I don't think we expect unioning objects on different workplanes to fail, right?). So now my thought is to try to get `cube` and `cyl` from my first post into the same stack to try to union them that way. I've tried:
```
import cadquery as cq

cube = cq.Workplane("XY").box(10, 10, 3)
cyl = cq.Workplane("XY").circle(5).extrude(3).translate((5, 0, -1.5))

new_stack = cq.Workplane("XY")
new_stack = new_stack.add(cube.all())
new_stack = new_stack.add(cyl.all())
```
but `new_stack` seems to be empty. Not sure what I've done wrong there!

jmwright, 2020-01-03:
@greyltc You should not be required to do them in the same stack, I just thought it would give some additional insight to make sure the union worked as expected. @adam-urbanczyk has marked this as a CAD kernel issue, and lately an indicator of a kernel issue has been having to add `clean=False`. We're working on moving to a newer version of the OCCT CAD kernel.

> but new_stack seems to be empty. Not sure what I've done wrong there!

I'd have to dig back through some code to see how adding objects to the stack should work. I typically try to stay away from manipulating the stack directly.

Is there a reason for not doing the following to create your model?
```python
import cadquery as cq

cube = cq.Workplane("XY").box(10, 10, 3).faces("<Z").workplane(offset=-1.5).center(5, 0).circle(5).extrude(3)

show_object(cube)
```
You should be able to call `combine` at the end of the chain if the objects are not fused the way you expect.

greyltc, 2020-01-03:
In this toy problem there's no reason not to do things all on one stack. I just created this code to illustrate something I saw in practice in a much more complicated operation.

I'm a bit new to cadquery and in practice, I've found it easier to keep track of things when I'm doing a long series of manipulations if I split them into separate workplanes sometimes. Probably as I get better at this, I'll not need to do that any longer. Though I do find that it can make my code a bit more readable and easier to debug when there aren't long lines of chained commands.

jmwright, 2020-01-03:
@greyltc 

> I just created this code to illustrate something I saw in practice in a much more complicated operation.

Thanks. It's good to have examples when things don't work as expected.

> I'm a bit new to cadquery and in practice, I've found it easier to keep track of things when I'm doing a long series of manipulations if I split them into separate workplanes sometimes.

Below are a couple of methods that are used to improve the readability of CadQuery code.

1. You could break the logical chunks of code up with variables to improve readability, something like this:
```python
import cadquery as cq

# the base of the part
cube = cq.Workplane("XY").box(10, 10, 3)

 # new workplane embedded in the base with the center changed for the boss
new_wp = cube.faces("<Z").workplane(offset=-1.5).center(5, 0)

# putting it all together
final_object = new_wp.circle(5).extrude(3)

show_object(final_object)
```
2. Here's a different way to make the code more readable:
```python
import cadquery as cq

cube = (cq.Workplane("XY").box(10, 10, 3) # the base of the part
                         .faces("<Z").workplane(offset=-1.5) # new workplane embedded in the base
                         .center(5, 0) # positioning the boss
                         .circle(5).extrude(3))

show_object(cube)
```

greyltc, 2020-01-03:
Thank you for the tips!

adam-urbanczyk, 2020-06-04:
With 7.4 I get:
```
Standard_ConstructionError: gp_Dir::Crossed() - result vector has zero norm
```