CadQuery/cadquery #264: Export to glTF

Peque, 2020-01-02:
Initially suggested in https://github.com/CadQuery/cadquery/issues/105.

JSON (I think that is what is called `TJS` in CadQuery) is deprecated in Three.js, and other formats seem to be lossy formats or 2D formats (like SVG).

As @fragmuffin mentioned:

> `cqparts` also has a [GLTF exporter](https://cqparts.github.io/cqparts/doc/api/cqparts.codec.html#module-cqparts.codec.gltf), which is just awesome (GLTF I mean :wink:)

Maybe some code can be reused.

Peque, 2020-01-02:
Advantages:

- Lossless (I think?)
- Multiple parts allowed
- Multiple colors/materials/textures

Maybe it could be used as the preferred intermediate format in CQ-editor GUI, `jupyter-cadquery`, `sphinx-cadquery`... I think it would be great if all those tools could share more code by using the same exporter/loader/visualizer. Or at least share as much as possible.

adam-urbanczyk, 2020-01-02:
Yet another reason to update to a newer OCCT version: https://dev.opencascade.org/doc/refman/html/package_rwgltf.html

NB: STEP is the lossless and "industrial" format. I think GLTF is actually lossy.

Peque, 2020-01-02:
Maybe relevant (about industry standards for 3D transmission formats):

- http://www.perey.com/ARStandards/3D_Transmission_Ninth_AR_Standards.pdf
- https://www.khronos.org/gltf/

jmwright, 2021-01-30:
Cross-referencing issues: https://github.com/CadQuery/OCP/issues/8

BrandonEscamilla, 2021-04-20:
HI, 

Is the GLTF exporter still in the plan?

adam-urbanczyk, 2021-04-20:
Yes

PR #802 (2021-11-13): Implement exporting to Gltf
Will resolve #264
closed 2021-11-13 by adam-urbanczyk