CadQuery/cadquery #266: largestDimension() is weird

greyltc, 2020-01-03:
Why is it the diagonal of the bounding box times five?
https://github.com/CadQuery/cadquery/blob/74573fc3bb032321c50a9e817dbf806c2a286d53/cadquery/cq.py#L2020

jmwright, 2020-01-03:
I think that code has been around since the first commit of CadQuery 1.x, so I'm not sure why it's that way. The [test](https://github.com/CadQuery/cadquery/blob/3cd466a693ce1d7f084e5cd6ddee256fc28d020a/tests/test_cadquery.py#L1052) seems to account for it being that way.

@dcowden Do you remember why it's that way? Was it a work-around for something in FreeCAD?

adam-urbanczyk, 2020-02-04:
@jmwright  I think it was related to how cuts used to be make (i.e. with CSG ).

adam-urbanczyk, 2020-04-02:
Do you want to use `largestDimension` for something @greyltc ? Otherwise I'd like to close the issue.

greyltc, 2020-04-02:
Yeah. I wanted to use it to find the largest dimension of an object :wink: 
I ended up writing my own that's pretty similar. It returns an accurate value (barring some corner cases that I didn't bother to address) and lets the user decide to measure along X, Y or Z:

https://github.com/greyltc/mechanical/blob/master/toolbox/utilities.py#L144-L161

It's not a drop in replacement, but it gets the job done for me.

I'm not really sure why this should be closed though, besides it being annoyoing to have issues open :wink:. The function as it is today doesn't do what its namesake says it does so I'd imagine it could cause someone else problems down the road if it's left like this.

jmwright, 2020-04-02:
I think at one time I tried using `longestDistance` to set camera position, but ended up using the bounding box, and combining the extents of the bounding boxes of all the objects in the scene when there was more than one.

https://github.com/jmwright/cadquery-gui/blob/master/assets/js/viewer.js#L157

My preference would be to leave this open and determine an example case that shows how it should work. If we can determine the expected behavior I'll put together a PR when I get a chance.

greyltc, 2020-04-02:
We could mesh the objects and calculate the distance between all possible pairs of verticies and return the largest value of that list...

adam-urbanczyk, 2020-04-02:
@greyltc this sounds rather slow. There is OBB in opencascade https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_bnd_lib.html#aed61fe542529ccd72e7e8fdad61a0bbe we could use it.
The main question here what is the desired largest dimension definition OBB,AABB,minimum enclosing sphere...

NB : the original question you had was answered (why diag*5 --> historical reasons), that is why I wanted to close the issue.

greyltc, 2020-04-02:
> the original question you had was answered (why diag*5 --> historical reasons), that is why I wanted to close the issue.

You're right!

greyltc, 2020-04-02:
I personally think the diameter of the minimum enclosing sphere would be good!

jmwright, 2020-04-29:
Corrected in #317

closed 2020-04-29 by jmwright