CadQuery/cadquery #269: Can CadQuery run in parallel (faster)?

OlegJakushkin, 2020-01-21:
So I installed `cadquery`  from `conda` like so:
```
conda install  -c conda-forge -c CadQuery CadQuery::pythonocc-core 
conda update  -c conda-forge  -c CadQuery -y --all
```

And I want to create a sphere. I try this sample:
```
import cadquery as cq
from cadquery  import exporters

wire = cq.Workplane("XY").threePointArc((1.0, 1.0), (0.0, 2.0)).close()
sphere = wire.revolve()
```

Yet it takes forever to finish... with `htop` showing 1 core 100% utilization

[image]

What do I do wrong and are there any parallelization options in `cadquery`?

jmwright, 2020-01-21:
There are not currently any parallelization options, and it's something I've looked into only briefly.

Your example completes almost instantly for me. I see that you have an exporter import at the top of the code snippet. If you're using an exporter and the precision is set too high, it can cause CPU utilization issues.

Please also compare your conda setup against the readme: https://github.com/CadQuery/cadquery#getting-started

OlegJakushkin, 2020-01-21:
@jmwright  Thank you, `sphere.val().exportStl(file, precision=1e-3)`  solved infinit time issue=)

closed 2020-01-21 by OlegJakushkin