CadQuery/cadquery #271: Helix seems to fail

RubenRubens, 2020-01-24:
### Helix is shorter than expected

I'm running Windows with the CQ-editor 0.1 RC1. I am working on a spring. I have tested it with two aproaches. The first one, using the native helix; the second one, using a parametric curve. **The problem is that the height of the helix is less than it should be.** `height` is set to 4, however it seems to be something around 2.5. 

Maybe I just did a mistake along the way, but I'm feeling like it's a bug.

Here's the code. (The helix curve is a copy-paste from [this gist](https://gist.github.com/fragmuffin/266bee4babd9a25aea2d53c739144ee3) of @fragmuffin.)

```python
import cadquery as cq
from math import sin, cos, pi

r = 3 # Radius of the helix
p = 1.5 # Pitch of the helix
h = 4 # Height of the helix

# Helix [first aproach]. It seems to fail.
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
shape = cq.Wire.combine([wire])
helix = cq.Workplane("XY").newObject([shape])

# Helix [second aproach]. It works as expected.
# helix = cq.Workplane('XY').parametricCurve(
# 	lambda t : (
# 		r * cos(t * 2 * pi),
# 		r * sin(t * 2 * pi),
# 		p * t
# 	),
# 	start=0,
# 	stop=h/p
# )

# Final result. A circle sweeped along a helix.
result = (
	cq.Workplane('XZ')
	.center(r, 0)
	.circle(0.1)
	.sweep(helix, isFrenet=True)
)
```

adam-urbanczyk, 2020-01-30:
Looks like a bug. Thanks for reporting.

jmwright, 2020-04-04:
@RubenRubens This was fixed by @adam-urbanczyk in #311

PR #311 (2020-04-04): Helix fix
Will resolve #271
closed 2020-04-04 by jmwright