CadQuery/cadquery #272: Feature placement inaccuracy

kaolpr, 2020-01-26:
Hi, there must be something I'm doing wrong as it seems to me highly improbable to be actually cadquery bug, however I have no idea what I'm doing wrong.

Here's the thing:
```
s = cq.Workplane("XY").box(1000, 600, 18.0, (False, False, False))
s = s.faces(">X").workplane().center(0.0, 0.0).hole(9.0)
s = s.faces(">X").workplane().center(-150.0, 0.0).hole(9.0)
s = s.faces(">X").workplane().center(150.0, 0.0).hole(9.0)
s.val().exportStep("debug.step")
```
From my understanding `workplane()` creates a plane with coordinate system centered in the `CenterOfMass`, in this case center of the box side. It should hold for all of 3 holes. 

However, having measured the output document, it turned out that first hole is located 150 mm from the edge (OK), second is 300 mm (OK) but third is 450.894106 mm, what is not what I expect. 

What's the reason for that?

jmwright, 2020-01-26:
@kaolpr It's a common issue to have. By default the center of the workplane is the center of mass. When you start removing material (asymmetrically especially), things can get strange. Try the `projectedOrigin` `centerOption` of `workplane`. https://github.com/CadQuery/cadquery/blob/102c16c14e42b6acca484efbf017f9ac2cc75789/cadquery/cq.py#L307

kaolpr, 2020-01-26:
Oh! Now it's so obvious! Thanks!

closed 2020-01-26 by kaolpr