CadQuery/cadquery #273: Where can I find more details about extending cq?

KeithSloan, 2020-01-30:
Where can I find more details about extending cq? i.e own functions that also use PythonOCC

adam-urbanczyk, 2020-01-30:
You can find some information in the docs: https://cadquery.readthedocs.io/en/latest/extending.html . Otherwise just ask here. Out of curiosity: what are you interested in specifically?

KeithSloan, 2020-01-30:
I had a look and it seems to be offering out of line functions.

I would like to provide new types of objects. Implemented using PythonOCC and essentially hidden away, don't want to clutter up source window. Specifically I would like to develop some functions for GDML solids where the parameters are very close to the GDML specification

adam-urbanczyk, 2020-01-30:
What do you mean by out of line functions? Regarding clutter you can always define a module with your extensions or wrap CQ in an additional module. Here you can find our sister project that does the latter: https://github.com/cqparts/cqparts .

KeithSloan, 2020-01-31:
out of line I mean def functions that you then call.

Thanks for the pointer to cqparts.

I wouid like to implement something that allowed for display and export of GDML like the
following
`import cadquery
import cqgdml

v = gVol("world")
s = gBox(10,10,10)
m = gMaterial('Air')
o = gObject(s,m,position,rotation)
v.add(o)
v.export("/tmp/file.gdml")
~
I can see how to setup class for the drawable things but not sure how to make it so that a export
would output formatted GDML to a file.

Made a start at https://github.com/KeithSloan/CQ-GDML-Module                            `

jmwright, 2020-01-31:
If you haven't already, you might want to have a look through CQ's [exporters.py code](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/exporters.py#L50) to see how different formats are converted and written to disk now.

KeithSloan, 2020-02-09:
Struggling with best way to develop.

I assume the ultimate aim would be to create an installable python module.

I am trying to develop with the definitions in another file i.e.
`import cadquery
import cqgdml

position = 0
rotation = 0

v = gVol("world")
s = gBox(10,10,10)
m = gMaterial('Air')
o = gObject(s,m,position,rotation)
v.add(o)
v.export("/tmp/file.gdml")
`

But it is not finding the definitions in cqgdml.py

adam-urbanczyk, 2020-02-09:
Maybe you wanted to do this:
```python
from cqgdml import *
```
i.s.o.
```python
import cqgdml
```

KeithSloan, 2020-02-15:
Okay

I now have the situation where I get an error NameError name show_object not defined.

What do I have to import for things to find show_object the examples on the web don't seem to require an import see https://github.com/KeithSloan/CQ_GDML_Package

jmwright, 2020-02-15:
The CQGI does some implicit import magic that's not one of the community's favorite features, but it's designed to make the scripts more portable between execution environments.

https://cadquery.readthedocs.io/en/latest/cqgi.html#the-execution-environment-side

KeithSloan, 2020-02-16:
I looked at the doc, but still don't know how to do 

`shape = cq.Workplane('XY').box(self.x,self.y,self.z)
  show_object(shape)`

In an imported python class

adam-urbanczyk, 2020-02-16:
Essentially you need to define `show_object` yourself (or run it in some sort of execution environment like CQ-editor).

KeithSloan, 2020-02-16:
Well I am running under cq-editor and it barfs that the function is not defined.

KeithSloan, 2020-02-19:
Okay from the imported module I loose access to show_object in the "parent" cq-editor cose/script.

Is there anyway round this?

adam-urbanczyk, 2020-02-20:
I'm not sure if you be doing such things, but the following seems to work:
```python
import sys
sys.modules['builtins'].__dict__['show_object'] = show_object
```
I'd personally only generate the object in modules and call `show_object` in the top level function only.

KeithSloan, 2020-02-20:
Thanks but did not work for me.

Tried print(sys.modules['builtins'].__dict__)

and got 
`{'__name__': 'builtins', '__doc__': "Built-in functions, exceptions, and other objects.\n\nNoteworthy: None is the `nil' object; Ellipsis represents `...' in slices.", '__package__': '', '__loader__': <class '_frozen_importlib.BuiltinImporter'>, '__spec__': ModuleSpec(name='builtins', loader=<class '_frozen_importlib.BuiltinImporter'>), '__build_class__': <built-in function __build_class__>, '__import__': <built-in function __import__>, 'abs': <built-in function abs>, 'all': <built-in function all>, 'any': <built-in function any>, 'ascii': <built-in function ascii>, 'bin': <built-in function bin>, 'breakpoint': <built-in function breakpoint>, 'callable': <built-in function callable>, 'chr': <built-in function chr>, 'compile': <built-in function compile>, 'delattr': <built-in function delattr>, 'dir': <built-in function dir>, 'divmod': <built-in function divmod>, 'eval': <built-in function eval>, 'exec': <built-in function exec>, 'format': <built-in function format>, 'getattr': <built-in function getattr>, 'globals': <built-in function globals>, 'hasattr': <built-in function hasattr>, 'hash': <built-in function hash>, 'hex': <built-in function hex>, 'id': <built-in function id>, 'input': <built-in function input>, 'isinstance': <built-in function isinstance>, 'issubclass': <built-in function issubclass>, 'iter': <built-in function iter>, 'len': <built-in function len>, 'locals': <built-in function locals>, 'max': <built-in function max>, 'min': <built-in function min>, 'next': <built-in function next>, 'oct': <built-in function oct>, 'ord': <built-in function ord>, 'pow': <built-in function pow>, 'print': <built-in function print>, 'repr': <built-in function repr>, 'round': <built-in function round>, 'setattr': <built-in function setattr>, 'sorted': <built-in function sorted>, 'sum': <built-in function sum>, 'vars': <built-in function vars>, 'None': None, 'Ellipsis': Ellipsis, 'NotImplemented': NotImplemented, 'False': False, 'True': True, 'bool': <class 'bool'>, 'memoryview': <class 'memoryview'>, 'bytearray': <class 'bytearray'>, 'bytes': <class 'bytes'>, 'classmethod': <class 'classmethod'>, 'complex': <class 'complex'>, 'dict': <class 'dict'>, 'enumerate': <class 'enumerate'>, 'filter': <class 'filter'>, 'float': <class 'float'>, 'frozenset': <class 'frozenset'>, 'property': <class 'property'>, 'int': <class 'int'>, 'list': <class 'list'>, 'map': <class 'map'>, 'object': <class 'object'>, 'range': <class 'range'>, 'reversed': <class 'reversed'>, 'set': <class 'set'>, 'slice': <class 'slice'>, 'staticmethod': <class 'staticmethod'>, 'str': <class 'str'>, 'super': <class 'super'>, 'tuple': <class 'tuple'>, 'type': <class 'type'>, 'zip': <class 'zip'>, '__debug__': True, 'BaseException': <class 'BaseException'>, 'Exception': <class 'Exception'>, 'TypeError': <class 'TypeError'>, 'StopAsyncIteration': <class 'StopAsyncIteration'>, 'StopIteration': <class 'StopIteration'>, 'GeneratorExit': <class 'GeneratorExit'>, 'SystemExit': <class 'SystemExit'>, 'KeyboardInterrupt': <class 'KeyboardInterrupt'>, 'ImportError': <class 'ImportError'>, 'ModuleNotFoundError': <class 'ModuleNotFoundError'>, 'OSError': <class 'OSError'>, 'EnvironmentError': <class 'OSError'>, 'IOError': <class 'OSError'>, 'EOFError': <class 'EOFError'>, 'RuntimeError': <class 'RuntimeError'>, 'RecursionError': <class 'RecursionError'>, 'NotImplementedError': <class 'NotImplementedError'>, 'NameError': <class 'NameError'>, 'UnboundLocalError': <class 'UnboundLocalError'>, 'AttributeError': <class 'AttributeError'>, 'SyntaxError': <class 'SyntaxError'>, 'IndentationError': <class 'IndentationError'>, 'TabError': <class 'TabError'>, 'LookupError': <class 'LookupError'>, 'IndexError': <class 'IndexError'>, 'KeyError': <class 'KeyError'>, 'ValueError': <class 'ValueError'>, 'UnicodeError': <class 'UnicodeError'>, 'UnicodeEncodeError': <class 'UnicodeEncodeError'>, 'UnicodeDecodeError': <class 'UnicodeDecodeError'>, 'UnicodeTranslateError': <class 'UnicodeTranslateError'>, 'AssertionError': <class 'AssertionError'>, 'ArithmeticError': <class 'ArithmeticError'>, 'FloatingPointError': <class 'FloatingPointError'>, 'OverflowError': <class 'OverflowError'>, 'ZeroDivisionError': <class 'ZeroDivisionError'>, 'SystemError': <class 'SystemError'>, 'ReferenceError': <class 'ReferenceError'>, 'MemoryError': <class 'MemoryError'>, 'BufferError': <class 'BufferError'>, 'Warning': <class 'Warning'>, 'UserWarning': <class 'UserWarning'>, 'DeprecationWarning': <class 'DeprecationWarning'>, 'PendingDeprecationWarning': <class 'PendingDeprecationWarning'>, 'SyntaxWarning': <class 'SyntaxWarning'>, 'RuntimeWarning': <class 'RuntimeWarning'>, 'FutureWarning': <class 'FutureWarning'>, 'ImportWarning': <class 'ImportWarning'>, 'UnicodeWarning': <class 'UnicodeWarning'>, 'BytesWarning': <class 'BytesWarning'>, 'ResourceWarning': <class 'ResourceWarning'>, 'ConnectionError': <class 'ConnectionError'>, 'BlockingIOError': <class 'BlockingIOError'>, 'BrokenPipeError': <class 'BrokenPipeError'>, 'ChildProcessError': <class 'ChildProcessError'>, 'ConnectionAbortedError': <class 'ConnectionAbortedError'>, 'ConnectionRefusedError': <class 'ConnectionRefusedError'>, 'ConnectionResetError': <class 'ConnectionResetError'>, 'FileExistsError': <class 'FileExistsError'>, 'FileNotFoundError': <class 'FileNotFoundError'>, 'IsADirectoryError': <class 'IsADirectoryError'>, 'NotADirectoryError [... cut ...] oftware Foundation.
All Rights Reserved.

Copyright (c) 2000 BeOpen.com.
All Rights Reserved.

Copyright (c) 1995-2001 Corporation for National Research Initiatives.
All Rights Reserved.

Copyright (c) 1991-1995 Stichting Mathematisch Centrum, Amsterdam.
All Rights Reserved., 'credits':     Thanks to CWI, CNRI, BeOpen.com, Zope Corporation and a cast of thousands
    for supporting Python development.  See www.python.org for more information.

Will have to look into returning shape to top level`

adam-urbanczyk, 2020-02-20:
What does not work @KeithSloan ? You need to run the code I pasted in your top level CQ script. I do not get why are you printing  `sys.modules['builtins'].__dict__`

KeithSloan, 2020-02-21:
Okay I am trying to now return things to the top level.
Ultimately I want a number of functions that add to the stack
and finally process the stack and return a shape to be shown.

For now I just want to test I can add a single object, and return it.
which has 

v = gVol("world")
show_object(v.shape2show())

and gVol has the following function
def shape2show(self) :
          import cadquery as cq
          cq.add(cq.Workplane('XY').box(10,10,10))
          #cq.Workplane.combine(self)
          return(cq.first(self))
          #return(cq.Workplane('XY').box(10,10,10))

If I return the commented return I get a box, but it barfs
cqquery has no attribute add where as this page seems to imply it is available~
https://pythonhosted.org/cadquery/apireference.html

adam-urbanczyk, 2020-02-22:
I think you are misinterpreting the docs. This is roughly what you want:
```python
import cadquery as cq

box1 = cq.Workplane('XY').box(1,1,1)
box2 = cq.Workplane('XY').transformed(offset=(0,0,2)).box(1,1,1)
combined = box1.add(box2)
show_object(combined)
```

NB: `cq.CQ` is a superclass of `cq.Workplane` and you were calling a (non-existing) function `add` from module `cq`.

KeithSloan, 2020-02-22:
Well I would like to access the stack as documented
"CadQuery methods that operate on the stack" under https://pythonhosted.org/cadquery/apireference.html

For GDML I would need to have a number of functions create shapes.
I was hoping to add them to the stack
and when I need to display access the stack, combine and return the highest level for show_object

adam-urbanczyk, 2020-02-22:
Does the snippet not work for you? It is combining multiple objects  using the `add` method just like you intended (NB: you can chain the `add` calls). Those objects could be created by your GDML module or any other means.

KeithSloan, 2020-02-23:
I am sure it works but using the stack would work better. Is the documentation on the stack not correct?

adam-urbanczyk, 2020-02-23:
> I am sure it works but using the stack would work better. Is the documentation on the stack not correct?

In the example I used the method `add` as it is documented and the resulting object is of type `cq.Workplane` which is a subclass of `cq.CQ`. In what sense would the documentation be not correct?

KeithSloan, 2020-02-23:
How do I specify/declare a vec as used in translate?

KeithSloan, 2020-02-23:
Okay I had everything working nicely but now I get an error cq-editor does not find my python include file which is in the directory from where I start cq-editor. as cqgdml.py

I get error ModuleNotFoundError cqgdml

And my code looks like

`import cadquery as cq

# need for reload
import importlib
#import cqgdml
import cqgdml
importlib.reload(cqgdml)
from cqgdml import *

rotation = 0

#display, start_display, add_menu, add_function_to_menu = init_display()

v = gVol("world")
s = gBox("box1",10,10,10)
m = gMaterial('G4_AIR0x55d123d17ea0')
o = gObject(s,m,None,rotation)
v.addObject(o)
show_object(v.shape2show())
v.exportVol("/tmp/file.gdml")`

Any ideas? I tried a reboot.

KeithSloan, 2020-02-24:
Is there a way to make an object display as wireframe?

jmwright, 2020-02-24:
@KeithSloan If you're using CQ-editor, look for the following button in the toolbar.

[image]

KeithSloan, 2020-02-24:
Thanks - How about under program control?

adam-urbanczyk, 2020-02-24:
It is not implemented. Can I actually close this issue @KeithSloan ? Are your questions regarding extending CQ answered?

KeithSloan, 2020-02-25:
Thanks - How about under program control? Yes you can close. My questions are really more support than issues

closed 2020-04-02 by adam-urbanczyk

tanius, 2021-01-11:
> Is there a way to make an object display as wireframe? […] How about under program control?

I accidentally found out that this can be done in CQ-editor by rendering an object with full transparency. You can use one of these options:

```
show_object(part, options = {"color": (64, 164, 100), "alpha": 1})  # preferred syntax
show_object(part, options = {"rgba": (64, 164, 100, 1)})  # mostly for backward compatibility
```

The only difference between this and the "real" wireframe mode available with the toolbar button in CQ-editor is that a fully transparent object will only show the helper wires in the face center when selected, while a real wireframe object will show them always. The best available  documentation of the `show_object()` syntax is [its test](https://github.com/CadQuery/CQ-editor/pull/127/commits/58e16d3).