CadQuery/cadquery #274: Which version of OCC does cadquery rely on ?

yguel, 2020-02-01:
Hi,
My goal is to test mouldability of some piece and if possible to correct them to make them mouldable.
The idea is to generate the mould and sweep it along its ejection path to obtain a solid X. One necessary condition for the piece to be mouldable is that there is no intersection between the piece and X.

One solution to get that is to make the Minkowski sum between the mould and the ejection path.

I would love to have a Minkowski operator available with cadquery but as it is a very complex problem, with researchers still working on it, I would like to make baby steps and first restricting the problem to ejection path being a segment is good enough for my problem.

This kind of Minkowski sum is simpler to implement, and I think it is a core feature that should be implemented in the geometric modeling engine. 

So I am thinking of implementing it in OCC and not in pythonocc nor in cadquery but having it available from cadquery.

I think the pythonocc version used by cadquery is the one by tpaviot (can someone confirm it or pointing to me how to check that info).
But I do not know which version of occ is used: oct, oce ...
Can somebody enlight me about that part ?

- all the best,

jmwright, 2020-02-01:
@yguel It's OCE 0.18.x

https://github.com/CadQuery/cadquery/blob/master/conda/meta.yaml#L16

tpaviot is in the process of updating PythonOCC to OCCT 7.4.x, and @adam-urbanczyk has been working on auto-generated bindings for OCCT 7.3.x and newer.

I think that we're headed towards using OCCT at some point, so it's probably best to contribute your work there of you want it in CadQuery at some point.

PythonOCC update: https://github.com/CadQuery/pythonocc-core/pull/2

Adam's Generator: https://github.com/CadQuery/pywrap

Generated Bindings: https://github.com/CadQuery/OCP

https://dev.opencascade.org/index.php?q=home/get_involved

yguel, 2020-02-01:
@jmwright Thanks a lot for the detailed answer.

I will ask tpaviot for his advice about OCE vs OCCT. One of his posts made me wonder if OCC/OCCT was still actively supported.
Nevertheless I will fill the form to be able to contribute directly to OCCT, it is very probable that the code necessary for the Minkowski sum will be compatible with OCE and OCCT for some time.

jmwright, 2020-02-01:
@yguel Sounds good. Please keep us updated on your progress.

adam-urbanczyk, 2020-02-02:
@yguel interesting problem. We will sooner or later move to OCCT7.4. Regarding PythonOCC the latest and greatest version (https://github.com/tpaviot/pythonocc-core/releases) is also based on OCCT7.4. So if I were you I'd  base my efforts on OCCT7.4.

closed 2020-04-02 by adam-urbanczyk