CadQuery/cadquery #276: Implement lightweight container for assemblies

adam-urbanczyk, 2020-02-04:
* Place existing parts in an assembly
* Allow to specify color/metadata per part
* Export correct STEP assemblies

adam-urbanczyk, 2020-02-04:
Based on feedback from FOSDEM

jmwright, 2020-02-04:
cqparts encompassed some of this functionality back in CQ 1.x.

adam-urbanczyk, 2020-02-04:
Indeed, it would be good to avoid duplication. I did mention CQparts in the Q&A, but it might be still useful to have such functionality (especially with good STEP R/W capabilities). Any thoughts @fragmuffin ?

bernhard-42, 2020-02-13:
There is also an Assembly/Parts hierarchy in jupyter-cadquery:
- Assembly( [<Part>*, <Assembly>* ], <name>)
- Part(<Shape>, <name>, <color>)

I needed it mainly to logically group objects together for visualisation and added attributes like show_faces and show_edges to control the initially visible parts of an assembly.

cqparts, afaik, added attributes and logic to really assemble Parts to one final object and is much more sophisticated.

crvio, 2020-02-17:
How would you recommend to currently assemble parts in new models?
(Meanwhile this is implemented sometime)
Is cqparts usable with CadQuery 2.0?
Do you think it would be wiser to go the jupyter-cadquery way?
Im just getting started with CadQuery (if not yet evident).

Sorry if posting in the wrong place, but I couldn't find a place to interact elsewhere.
If so, please just point it out.
Thanks in advance.

jmwright, 2020-02-17:
@crvio 

> Is cqparts usable with CadQuery 2.0?

It's not, but I'm working on it. https://github.com/cqparts/cqparts/issues/129#issuecomment-586663638

Even when that fork is ready, we'll need some other changes to make it into master on this repo before it will be fully usable.

> Do you think it would be wiser to go the jupyter-cadquery way?

I haven't used assemblies or parts in jupyter-cadquery, but it's being actively developed by @bernhard-42 . cqparts is not being actively developed at this time, but I'm hoping that updating it for CQ 2.0 will help jumpstart that process again. If I were you I would give jupyter-cadquery a try, and keep an eye on cqparts to see how things with it work out.

adam-urbanczyk, 2020-02-23:
@crvio in the meantime you can use the following (somewhat primitive) apporach in pure CQ:
```python
import cadquery as cq

box1 = cq.Workplane('XY').box(1,1,1)
box2 = cq.Workplane('XY').transformed(offset=(0,0,2)).box(1,1,1)
box3 = cq.Workplane('XY').transformed(offset=(0,1,1)).box(1,1,1)

res = box1.add(box2).add(box3)
```
resulting in
[image]

crvio, 2020-02-24:
thanks @adam-urbanczyk , will do

Shimwell, 2020-02-28:
Fab this is just what I have been looking for

PR #440 (2020-10-01): Assembly support
This PR will eventually resolve #276, #20

What's left

- [x] ~~Exact jacobian~~ , finite diff jacobian by hand
- [x] Finalize solver choice -> BFGS it is
- [x] Implement parameters handling (e.g. specific angle or distance constraint)
- [x] Visually attractive example in docs/README
closed 2020-10-01 by jmwright