CadQuery/cadquery #277: Sweep from face of extrusion fails

bragostin, 2020-02-06:
Why does this work
```python
import cadquery as cq
PDI = 4.
PDO = 8.
PDL = 20.
path = cq.Workplane('YZ').moveTo(0,PDL/3).lineTo(0,2*PDL/3)
port = cq.Workplane('XY').workplane(offset=PDL/3).circle(PDO/2).workplane(offset=PDL/3).circle(PDI/2).sweep(path, multisection=True
show_object(port))
```
and this does not?
```python
import cadquery as cq
PDI = 4.
PDO = 8.
PDL = 20.
s_shell = cq.Workplane('XY').box(8, 122, 62)
port = s_shell.faces('>Z').circle(PDO/2).extrude(PDL/3)
path = cq.Workplane('YZ').moveTo(0,PDL/3).lineTo(0,2*PDL/3)
port = port.faces('>Z').workplane().circle(PDO/2).workplane(offset=PDL/3).circle(PDI/2).sweep(path, multisection=True)
show_object(port)
```
anything obvious I missed here?

adam-urbanczyk, 2020-02-06:
I get a segfault, do you get one as well? Here is a workaround for now (NB: loft i.s.o. sweep):
```python
import cadquery as cq
PDI = 4.
PDO = 8.
PDL = 20.
s_shell = cq.Workplane('XY').box(8, 122, 62)
port = s_shell.faces('>Z').circle(PDO/2).extrude(PDL/3)
port = port.faces('>Z').workplane().circle(PDO/2).workplane(offset=PDL/3).circle(PDI/2).loft()
```
I inspected the object in CQ-editor and the path does not start when the staring face is located. It might be the root cause, but I still need to confirm.

bragostin, 2020-02-06:
@adam-urbanczyk yes I get a segfault. ``loft`` works fine in this case since the extrusion path is straight. Thank you!

adam-urbanczyk, 2020-02-13:
Here is a solution. I think it should be prettier, feel free to share your ideas regarding our API improvements.

```python
import cadquery as cq

PDI = 4.
PDO = 8.
PDL = 20.

s_shell = cq.Workplane('XY').box(8, 122, 62)
port = s_shell.faces('>Z').circle(PDO/2).extrude(PDL/3)
path = cq.Workplane('YZ',origin=port.faces('>Z').val().Center()).lineTo(0,PDL/3)
port = (port.faces('>Z').workplane()
            .circle(PDO/2).workplane(offset=PDL/3)
            .circle(PDI/2)
            .sweep(path,multisection=True))
```

I think that the spine was misplaced w.r.t. the two circles of interest.

bragostin, 2020-02-14:
@adam-urbanczyk thank you, that works fine!
Following the same logic as your solution, if everything is parametrized the code below works also
```python
import cadquery as cq
PDI = 4.
PDO = 8.
PDL = 20.
H = 8.
W = 122.
L = 62.
s_shell = cq.Workplane('XY').box(H, W, L)
port = s_shell.faces('>Z').circle(PDO/2).extrude(PDL/3)
path = cq.Workplane('YZ').center(0,L/2+PDL/3).lineTo(0,PDL/3)
port = (port.faces('>Z').workplane()
            .circle(PDO/2).workplane(offset=PDL/3)
            .circle(PDI/2)
            .sweep(path,multisection=True))
show_object(port)
```
That does not explain the underlying reason for this origin shift though. I am trying to look into it to find out the reason, but no luck so far.

adam-urbanczyk, 2020-02-16:
I'm no sure that I get what you mean, the original "simple" code had the path aligned with the two sections:
[image]

bragostin, 2020-02-17:
@adam-urbanczyk yes you are right, I got confused.

Interestingly, if I run the following code 
```python
H = 8.
W = 122.
L = 62.
s_shell = cq.Workplane('XY').box(H, W, L)
port = s_shell.faces('>Z').circle(PDO/2).extrude(PDL/3)
port = port.faces('>Z').workplane()
print("port.plane.origin = ", port.plane.origin)
```
I get ``port.plane.origin =  Vector: (0, 0, 37.67)``, while if I place ``print("self.plane.origin = ", self.plane.origin)`` inside the ``def workplane(
        self, offset=0.0, invert=False, centerOption="CenterOfMass", origin=None
    ):`` of ``cq.py``, I get ``self.plane.origin =  Vector: (0.0, 0.0, 0.0)``
Shouldn't both be equal?