CadQuery/cadquery #278: FOSDEM 2020 - video online

adam-urbanczyk, 2020-02-06:
[Towards CadQuery 2.0](https://fosdem.org/2020/schedule/event/cadquery/)

dcowden, 2020-02-06:
I watched the video-- great job Adam!  It was very interesting to hear the questions.   

I'm shocked that people don't immediately see how much better CQ is that OpenSCAD-- you only have to look at the code for a model that has anything other than a square box for about 100ms to see!

I'm really excited about the possibility that we could get more contributors and really make some improvements.  

To my knowledge, there is nothing like CQ in existence.   I can see now that my ideas about growing CQ through the efforts of 'end users' (parametric parts) was probably incorrect.  The power is much more clear when you've got a business project like KiCAD, or the railroad toy folks,  and you need to dynamically engineer your models.  I can see it being introduced at the grass-roots level by engineers to make their lives easier, and eventually growing to be a tool that's used all over the place.

bragostin, 2020-02-06:
From my experience I can confirm that there is nothing like CQ in existence. I have worked with Openscad before and as my models grew more complex (with large patterns of small elements) I ran into limitations that I found impassable. CQ allowed me to overcome these easily I must say. Thank you all for this awesome work!

jmwright, 2020-02-06:
Nicely done @adam-urbanczyk ! I've been looking forward to watching this. 

Thank you for your willingness to speak at FOSDEM. It is very much appreciated.

dcowden, 2020-02-06:
@bragostin that's good to hear.  The decision to use python is key.  As the hands-down science and math winner, the sky's the limit for for what CQ can do.

closed 2020-05-02 by jmwright