CadQuery/cadquery #279: Loft faces of two solids?

r4dr3fr4d, 2020-02-11:
`two_boxes = cq.Workplane("XY").box(1,1,1).faces(">Z").workplane(offset=5).box(1,1,1).faces("<Z").loft()`

Have spent a while trying to make this work, but it complains of only having one wire.  I may not yet have my mind wrapped around how to properly use the stack; if I figure it out I'll post here.

jmwright, 2020-02-11:
I'm not sure what your intent is. A `box` is a 3D object. Did you mean to use `rect`s (2D objects) instead and loft between them?

https://github.com/CadQuery/cadquery/blob/master/examples/Ex018_Making_Lofts.py

Notice that in the example the loft is done between the circle and rect. The face of the box is just used for positioning the circle.

r4dr3fr4d, 2020-02-11:
I saw the example.  I have two boxes, next to each other but not overlapping.  I'd like to loft the space between the closest faces (e.g., "<Z" on one and ">Z" on the other) somehow.  Do I have to create another shape after selecting a face before lofting?  That seems redundant to me.

r4dr3fr4d, 2020-02-11:
I also tried appending .wires() to each .faces(), which works fine for selecting the wires I want to loft, but they aren't both picked up on the same stack - if that makes sense.

jmwright, 2020-02-11:
For some reason `loft` won't accept a wire from a selected face, you have to add it manually.
```python
ttwo_boxes = cq.Workplane("XY").box(1,1,1).faces(">Z").rect(1, 1).workplane(offset=5).box(1,1,1).faces(">Z").rect(1, 1).loft()
```
I can't think of a reason why right now. It seems intuitive that you could just do `faces(">Z").wires().first()`, but it doesn't work. Hopefully another community member will see what I'm missing.

jmwright, 2020-02-11:
Actually, I guess it's because `circle` and `rect` push their wires onto the stack, but I tried manipulating the stack to add the wires from the selected faces and it still did not work. I typically don't directly mess with the stack at all, so I'm pretty rusty on how it works. It seems like this might be a good use case for tagging objects, which was a feature that was recently merged, and is not included with CQ-editor yet: https://github.com/CadQuery/cadquery/pull/252

adam-urbanczyk, 2020-02-11:
I think it is a duplicate of #148 

Extrude,loft,... look for wires in `pendingWires`.

r4dr3fr4d, 2020-02-11:
I would think .wires() to be an acceptable solution; having to define the geometry of the face again would stink.  Furthermore, I haven't been able to hull 2x of an entire box or sphere either, I guess for the same reason.  It would make sense to me that you can also loft two solids.

adam-urbanczyk, 2020-02-13:
`wires()` is a selector, it has nothing to do with `pendingWires`. I don't think it would be a good idea to change semantics of such a basic  CQ method, hence the proposal in #148 . AFAIR at most faces are supported for sweep/loft in Opencascade which is an upper bound on what would be supported by CQ. NB: Opencascade uses B-Rep. I am not sure what you mean by "hull 2x".

r4dr3fr4d, 2020-02-14:
Sorry, I meant hulling two solids together, like two spheres.

r4dr3fr4d, 2020-02-17:
@adam-urbanczyk so I've read through the code and follow what you're saying - would it make sense to add an argument to loft() to use in lieu of pendingWires?  Or would a more appropriate solution be to have pendingWires appended to by faces()?  Not sure if there's a clear winner here.

r4dr3fr4d, 2020-02-17:
Made progress (sort of) with the extrude example on #148 using vals():

```
box1 = cq.Workplane("XY").box(1,1,1).faces(">Z")
box2 = cq.Workplane("XY", origin=(0,0,5)).box(1,1,1).faces("<Z")
two_boxes = box1.add(box2).wires()
two_boxes.ctx.pendingWires = two_boxes.vals()
two_boxes = two_boxes.loft()
```

The issue with this is, though the faces being lofted are the two "inner" ones of the boxes, facing each other, the original box (at (0,0,0) is included in the loft, instead of its one face.  Any thoughts as to why that is?

r4dr3fr4d, 2020-03-01:
in loft(), `combine` needs to be `False`.  Will close this since this issue is sort of a duplicate.

closed 2020-03-01 by r4dr3fr4d