CadQuery/cadquery #280: Local/global coordinate system problem

KAndrsn, 2020-02-15:
Example:  The goal is to create a curved handle and build a tool on the far end (because there is a hand grip on the near end).  I've simplified this to a simple rectangle cross-section.  I think part of my problem could be starting with plane "ZX", but I wanted to match my 3d printer's coordinate system and changing this causes things to break.

Attaching the extruded rect to the .faces() result.  The block is aligned to the global xyz, not the normal of the face. (Comment out the .workplane.transformed().)
[image]

Attach the extruded rect to the .faces().workplane() result.  The extrude is along the normal to the face, but "up" gets reset/forgotten(?).
[image]

So, I added the .transformed() call to rotate 90 degrees about the local y-axis, but instead get this, which is rotated about the global Y axis.
[image]

```import cadquery as cq
revolve_radius = 300
result = (cq.Workplane("ZX")
	.rect(10,20)
	.revolve(45.0, (1.0,revolve_radius), (-1.0,revolve_radius))
	)
result = (result
	.faces(">X and >Y")
	.workplane()
	.transformed(rotate=(0,90,0))
	)
result = (result
	.rect(10,20)
	.extrude(5)
	)
show_object(result)
```

bragostin, 2020-02-16:
I think this issue is the same as #254
There is a Pull Request in the making to correct this behaviour I believe.
I found this workaround using ``toWorldCoords`` to work in my case:
```python
pln = part.faces(face).workplane(centerOption='ProjectedOrigin', origin=(0,0,0))
rot = cq.Vector(pln.plane.toWorldCoords((ANGLE, 0, 0))) 
part = pln.transformed(rotate=rot).pushPoints(xy).rect(Length, Width).cutBlind(Lcut)
```

adam-urbanczyk, 2020-07-20:
This is what the code produces now. Is it the expected result @KAndrsn ?
[image]

KAndrsn, 2020-07-20:
The goal is for the 10x20x5 block to align seamlessly with the end of the revolved extrusion.  The code above may be incorrect for this.  (Note: This is a simplified example.  The idea is to have a curved I-beam handle, and build the "tool" on the end of the handle. The 10x20x5 block would be a transition shape from the I-beam to... whatever.)

adam-urbanczyk, 2020-08-04:
You can get it with the following code.
```python
revolve_radius = 300
result = (cq.Workplane("ZX")
	.rect(10,20)
	.revolve(45.0, (1.0,revolve_radius), (-1.0,revolve_radius))
	)
result = (result
	.faces(">X and >Y")
	.workplane()
	.transformed(rotate=(0,0,90))
	)
result = (result
	.rect(10,20)
	.extrude(5)
	)
show_object(result)
```
[image]

closed 2020-08-04 by adam-urbanczyk