CadQuery/cadquery #281: Investigate using VTK for web based visualization

adam-urbanczyk, 2020-02-16:
Current approach is based on x3dom functionality of python-occ. It will probably be not available in the future but it would be good to keep the simple ipython visuaization capabilities. One way to achieve this woulde be to use VTK and some additional tools. Relevant materials

* https://lorensen.github.io/VTKExamples/site/Python/DataManipulation/Cube.py
* https://k3d-jupyter.org/index.html
* https://docs.pyvista.org/index.html

Note that OCCT has (in some versions) VTK related functionalities:

* https://www.opencascade.com/doc/occt-7.0.0/refman/html/toolkit_tkivtk.html --> has it been removed in 7.4 ?

bernhard-42, 2020-03-15:
I agree searching for an alternative to x3dom is a good idea.
I had severe problems with x3dom when I built my first jupyter extension for cadquery. The biggest one was that with x3dom I regularly ran out of Webgl contexts, and I did not find a way to control the building of webgl contexts. And there is a lack of flexibility from a developer perspective.

That's why I switched over to pythreejs (and pythonocc-core also comes with a pythreejs based jupyter renderer). The overall experience with pythreejs is good. From a documentation perspective, I very often needed to rely on the three.js docs, but since pythreejs is a thin layer on top of threejs, most of the stuff works as expected.

adam-urbanczyk, 2021-04-09:
Some notes:

getting `vtkPolyData`:
```python
vs = OCP.IVtkOCC.IVtkOCC_Shape(w)
sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
sm.Build(vs,sd)

res = sd.getVtkPolyData()
```

So far nice simple visualziation with pyvista+itkwidgets. VTK.js being the lower-level lib
```python
mesh = pv.PolyData(res)
pv.plot_itk(mesh, smooth_shading=True)
```
[image]

adam-urbanczyk, 2021-04-11:
MWE for future reference:
```python
import OCP
import vtk
import pyvista as pv

shape  = cq.Workplane().box(1,1,1).faces('>Z').hole(0.7).edges().fillet(0.03)

vs = OCP.IVtkOCC.IVtkOCC_Shape( shape.val().wrapped )
sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
sm.Build(vs,sd)

res = pv.PolyData(sd.getVtkPolyData())
res.clear_arrays()

pv.plot_itk(res, smooth_shading=True)
```

Without clear_arrays I get and error form ITK widgets (a bug - seems to not like the Int64 datatype).

bernhard-42, 2021-04-11:
Thanks for the MWE.

Looks like `itkwidgets` could be a tricky dependency for cadquery: Besides having issues with the arrays it doesn't support the latest JupyterLab 3 at the moment (https://github.com/InsightSoftwareConsortium/itkwidgets/issues/405). 
And if the Jupyter folks continue delivering breaking major releases it gets a real effort for devs to always support the latest Jupyter Lab (already hit me twice with two projects) - while users ask for the latests versions soon after release.

Just for the records, the arrays are properly supported in the visualizer of IPython, so it is most probably on the itkwidgets level.

bernhard-42, 2021-04-11:
What I don't understand is why `n_faces` is not the length of the reshaped `faces` array:

```python
In [15]: import cadquery as cq
    ...: import OCP
    ...: import pyvista as pv
    ...:
    ...: shape  = cq.Workplane().box(1,2,3)
    ...:
    ...: vs = OCP.IVtkOCC.IVtkOCC_Shape( shape.val().wrapped )
    ...: sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
    ...: sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
    ...: sm.Build(vs,sd)
    ...:
    ...: res = pv.PolyData(sd.getVtkPolyData())
    ...: print(f"number of points: {res.n_points}")
    ...: print(f"number of normals: {res.face_normals.shape}")
    ...: print(f"number of faces (1): {res.n_faces}")
    ...: print(f"number of faces (2): {res.faces.shape}")
    ...: print("faces:")
    ...: for i, face in enumerate(res.faces.reshape(-1, 4)):
    ...:     print(f"{i:2}", face)
    ...:
number of points: 128
number of normals: (68, 3)
number of faces (1): 68
number of faces (2): (48,)
faces:
 0 [  3 105 104 106]
 1 [  3 105 106 107]
 2 [  3 109 108 110]
 3 [  3 109 110 111]
 4 [  3 115 113 112]
 5 [  3 115 112 114]
 6 [  3 119 117 116]
 7 [  3 119 116 118]
 8 [  3 123 121 120]
 9 [  3 123 120 122]
10 [  3 127 125 124]
11 [  3 127 124 126]
```

So there are 12 triangles in the array (which makes sense), and it seems to ignore 104 of the points (duplicates?). And even the used points 104...127 contain many duplicates (at the end we only have 8 vertices on a box).

And pyvista doesn't visualize the triangles, but 24 sub-faces as squares: 

[image]

I guess it all comes down to a correct interpretation of the `MESH_TYPES` and `SUBSHAPE_IDS` arrays, which don't seem to be documented extensively ...

Need to further look into it before being able to use this data to replace my tessellation.

adam-urbanczyk, 2021-04-11:
In the data I see two arrays:
```
pyvista DataSetAttributes
Association: CELL
Contains keys:
	SUBSHAPE_IDS
	MESH_TYPES
```
IDs just points to distinct entities (face,s egdes, wires)
TYPES denotes cell type and (I assume is based on this https://dev.opencascade.org/doc/occt-7.0.0/refman/html/_i_vtk___types_8hxx.html#a0c77217fdb907063f07ee89b2b0fe060:
```
MT_Undefined | Undefined.
MT_IsoLine | Isoline.
MT_FreeVertex | Free vertex.
MT_SharedVertex | Shared vertex.
MT_FreeEdge | Free edge.
MT_BoundaryEdge | Boundary edge (related to a single face)
MT_SharedEdge | Shared edge (related to several faces)
MT_WireFrameFace | Wireframe face.
MT_ShadedFace | Shaded face.
```

adam-urbanczyk, 2021-04-11:
PyVista displays all the cells that are present in the dataset - those include vertices, edges, isolines and faces. You are interested in the faces only (`MESH_TYPES==7` I think).  

I don't understand what else needs to be investigated to replace `cq.Shape.tessellate`. `res.points` and `res.faces` gives you exactly the same information, you just need to reshape `faces`.

Regarding ITK, I assume they'll fix it eventually. Another option would be to use vtk.js directly.

adam-urbanczyk, 2021-04-11:
Additional observations, conversion to float types solves the issue
```python
res.cell_arrays['SUBSHAPE_IDS'] = res.cell_arrays['SUBSHAPE_IDS'].astype(float)
res.cell_arrays['MESH_TYPES'] = res.cell_arrays['MESH_TYPES'].astype(float)
```

I guess that the issue has something to do with js not supporting int64.

bernhard-42, 2021-04-11:
Thanks a lot for the investigation, this is really helpful

> You are interested in the faces only 

I actually need faces, normals and edges. Inspired by Thomas' code in pythonocc, my latest [tessellator](https://github.com/bernhard-42/jupyter-cadquery/blob/master/jupyter_cadquery/tessellator.py) derives all of it based on the mesh created by `BRepMesh_IncrementalMesh` (which btw. allows parallel computing which is not supported by the vtk stuff as far as I see). 

> I don't understand what else needs to be investigated to replace cq.Shape.tessellate

Another important point is how to ensure to get triangles from the mesh. Since the format of each face is `(num_points, x0, x1, ...)` I would expect that the algorithm might also return faces that are not triangles. If it returns different face types, I cannot simply `reshape` the `faces` array but have to loop (which most probably is much slower than reshaping).

I will now try to derive what I need from vtk based on the types you found.

adam-urbanczyk, 2021-04-11:
The `vtkPolyData` object is produced by the OCCT tessellator (https://github.com/Open-Cascade-SAS/OCCT/blob/f7ad1e7e3729c128014a23f18c028c55f50db301/src/IVtkOCC/IVtkOCC_ShapeMesher.cxx#L133) and you'll only get triangles. Indeed the parallel option is not exposed. You can get edges like so `res.lines` and normals like so `res.cell_normals`.

bernhard-42, 2021-04-11:
I was actually just reading through this source file and found that it `IVtkOCC_ShapeMesher::addShadedFace` indeed triangulates. Glad you also confirmed that.

I can now create a tessellate function based on vtk and test/time it on my examples.

Thanks again for all your support. It would have taken me ages to get that far.

bernhard-42, 2021-04-11:
Note on parallelism: if you set `BRepMesh_IncrementalMesh.SetParallelDefault_s(True)` then the ShapeMesher runs parallel on all cores.
```text
In [14]: shape = cq.importers.importStep("pythonocc-demos/assets/models/RC_Buggy_2_front_suspension.stp")

In [15]: BRepMesh_IncrementalMesh.SetParallelDefault_s(False)

In [16]: %%time
    ...: vs = OCP.IVtkOCC.IVtkOCC_Shape( shape.val().wrapped )
    ...: sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
    ...: sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
    ...: sm.Build(vs,sd)
    ...:
    ...:
CPU times: user 13.5 s, sys: 176 ms, total: 13.7 s
Wall time: 13.6 s

In [17]: shape = cq.importers.importStep("pythonocc-demos/assets/models/RC_Buggy_2_front_suspension.stp")

In [18]: BRepMesh_IncrementalMesh.SetParallelDefault_s(True)

In [19]: %%time
    ...: vs = OCP.IVtkOCC.IVtkOCC_Shape( shape.val().wrapped )
    ...: sd = OCP.IVtkVTK.IVtkVTK_ShapeData()
    ...: sm = OCP.IVtkOCC.IVtkOCC_ShapeMesher()
    ...: sm.Build(vs,sd)
    ...:
    ...:
CPU times: user 30.2 s, sys: 1.28 s, total: 31.5 s
Wall time: 4.9 s
```

bernhard-42, 2021-04-17:
@adam-urbanczyk 

Some more VTK research update:

I built a tessellator based on VTK (see at the end). It looks promising until one needs to calculate the edges (deviation params 
 are selected to be the same as for the other tests):

```text
%%time
import time

t = time.time()
tess = VTKTessellator(rc)
tess.compute(deviation=0.0008, deviation_angle=0.3/3.1415*180, iso_lines=0, parallel=True)
print("compute", time.time() -t)

t = time.time()
vertices = tess.vertices
print("vertices", time.time() -t)

t = time.time()
triangles = tess.triangles
print("triangles", time.time() -t)

t = time.time()
normals = tess.normals
print("normals", time.time() -t)

t = time.time()
edges = tess.edges
print("edges", time.time() -t)
```

results in:

```text
0.953884261730799 0.29999999999999993
compute 2.1458945274353027
vertices 0.0008137226104736328
triangles 0.02211904525756836
normals 0.35484933853149414
edges 2.871659278869629

CPU times: user 18.6 s, sys: 537 ms, total: 19.2 s
Wall time: 5.4 s
```

Nearly 3 seconds to get the edges. Reason is the format of polygons: `(l1, ind_1_1, ..., ind_1_l1, l2, ind_2_1, ..., ind_s_l2, ...)` with `l1`, `l2`, ... being arbitrary integers depending on the length of the polygon. Hence, a simple reshaping doesn't work, one  has to loop over the array and extract every single sub array of indexes and map the indexes to point coordinates.
Up to now I haven't found a performant way in numpy to get this to the required format (a list of simple 2-vertex edges)

```python
[
    [(x1,y1,z1),(x2,y2,z2)],
    [(x3,y3,z3),(x4,y4,z4)],
    ...
]
```

So the current situation for the RC car example is:

- C++ tesselator as of pythonocc: 2.9 sec
- VTK tesselator: 5.4 sec
- Optimized Python tessellator: 9.4 sec

I will continue to try to find a faster way to get the edges. This is independent of the underlying OCP since I only need to transform the numpy array I get from the VTK mesher.

For completeness, I add my current VTKTesselator (still work in progress):

```python
from enum import Enum
[... 99 lines cut ...]
                MeshType.BoundaryEdge.value: array("f"),
                MeshType.SharedEdge.value: array("f"),
                MeshType.SeamEdge.value: array("f"),
            }

            while offset < lines.shape[0]:
                l = lines[offset]
                line = self._points[lines[offset+1:offset+1+l]]
                edges[types[i]].extend(np.hstack((line[:-1], line[1:])).reshape(1,-1)[0])
                
                offset += l + 1
                i = i + 1

            self._edges = {}
            for mt in edges.keys():
                self._edges[MeshType(mt)] = np.asarray(edges[mt]).reshape(-1, 2, 3)
            
        return self._edges
    
    @property
    def normals(self):
        if self._normals is None:
            self._normals = np.asarray(self._poly_data.point_normals, dtype=np.float32)
        return self._normals
```

adam-urbanczyk, 2021-04-18:
The following code seems to convert the lines to have always length of 2. You can then use a simple numpy reshape.
```python
def polylineToMultiline(source):
    triangles = vtk.vtkTriangleFilter()
    triangles.SetInputData(source)
    triangles.Update()
    return pv.PolyData(triangles.GetOutput())
```
Based on: https://discourse.vtk.org/t/how-can-i-split-a-polyline-into-its-linear-segments/2434/5

bernhard-42, 2021-04-18:
Nice, let me try. I also found a way to really speed up the segment creation.

I am now down to 0.65 sec from 2.87 sec (just forget about all fancy array and numpy stuff and handcraft the copy action)

Note, I have renamed "edges" to "segments"
```text
deflection= 0.95388, angular deflection= 0.30000, parallel=True
compute  : 2.0992343425750732
triangles: 0.04112362861633301
normals  : 0.4001476764678955
segments : 0.6497976779937744
CPU times: user 16.3 s, sys: 782 ms, total: 17.1 s
Wall time: 3.19 s
```

And, woohhoo, we are with 3.19 sec in the range of pythonocc, goal achieved!

I will now check whether your `polylineToMultiline` approach will further improve the results.

adam-urbanczyk, 2021-04-18:
BTW: are you sure that your normals shouldn't be `cell_normals` + selection based on subshape id?

bernhard-42, 2021-04-18:
> The following code seems to convert the lines to have always length of 2. You can then use a simple numpy reshape.

This works very well (the overall rendering time is now down to 2.8 sec) and simplifies the code, great finding!

I still keep the line mesh  types separate, so that one can e.g. provide different colors to iso lines than to other lines.

```python
    def segments(self):
        segments = self._poly_data.lines.reshape(-1,3)[:,1:]
        line_types = tess.mesh_types[self.num_vertices:self.num_vertices + self.num_segments]
        typed_segments = np.column_stack((line_types, segments))

        self._segments = {}
        for key in [MeshType.FreeEdge, MeshType.SharedEdge, MeshType.IsoLine, MeshType.BoundaryEdge, MeshType.SeamEdge]:
            self._segments[key] = self._points[typed_segments[np.in1d(typed_segments[:, 0], np.asarray([key.value]))][:,1:]]
```

> BTW: are you sure that your normals shouldn't be cell_normals + selection based on subshape id?

I wasn't sure. However by pushing all points and all point_normals to pythreejs the objects got visualised which together with performance, was the initial goal.
Now that the performance is OK, I will check how to properly work with normals.

bernhard-42, 2021-04-18:
> BTW: are you sure that your normals shouldn't be cell_normals + selection based on subshape id?

As can be seen from https://threejsfundamentals.org/threejs/lessons/threejs-custom-buffergeometry.html, threejs expects *point normals*. 

I didn't find a VTK'ish way to determine the offset in the point array for points belonging to triangles, so I chose to simply use the min index value. Goal is to only push points to pythreejs that are actuall needed for the rendering to minimize memory usage in the browser. This depends on the implementation details of the VTK tesselator that adds faces (triangles) as the last step.

My latest version now is:

```python
from enum import Enum
from OCP.TopExp import TopExp_Explorer
from OCP.TopAbs import TopAbs_SOLID

class MeshType(Enum):
    # Order of cell blocks from OCP vtk:
    # - Vertices:   1/FreeVertex, 2/Shared_Vertex (will be mixed)
    # - Edges:      0/IsoLine (only when requested), 3/FreeEdge, 4/BoundaryEdge, 5/SharedEdge, 8/SeamEdge (will be mixed)
    # - WireFrames: 0/IsoLine
    # - Faces:      7/ShadedFace

    IsoLine = 0        # Isoline                                                                 
    FreeVertex = 1     # Free vertex                                                             
    SharedVertex = 2   # Shared vertex                                                           
    FreeEdge = 3       # Free edge                                                               
    BoundaryEdge = 4   # Boundary edge (related to a single face)                                
    SharedEdge = 5     # Shared edge (related to several faces)                                  
    WireFrameFace = 6  # Wireframe face                                                          
    ShadedFace = 7     # Shaded face                                                             
    SeamEdge = 8       # Seam edge between faces                                              

class VTKTessellator:
    def __init__(self, shape):
        self.shape = shape
        self.parallel = self._multi_shape()
        self._poly_data = None
        self._points = None

        self._vertices = None
        self._triangle_points = None
        self._triangle_point_normals = None
        self._segments = None
        self._mesh_types = None
        self._subshape_ids = None

        self.num_vertices = self.vertices_offset = None
        self.num_segments = self.segments_offset = None
        self.num_triangles = self.triangles_offset = None
        
    def _multi_shape(self):
        """
        Check whether a shape or compound has multiple subshapes
        
        :param shape: An OCP.TopoDS.TopoDS_Compound or OCP.TopoDS.TopoDS_Solid
        :returns: True for multiple subshapes, False else
        """
        
        count = 0
        e = TopExp_Explorer(self.shape, TopAbs_SOLID)
        while e.More() and count <= 1:
            count += 1
            e.Next()
        return count > 1
        
    def compute(self, deviation=0.0001, deviation_angle=12, iso_lines=0, parallel=None, info=False):
        """
        Tessellate the shape
        :param deviation: Multiplier for the internally calculated deflection value (default=0.0001)
        :param deviation_angle: Angle in degree for the angular deflection (default=12)
        :param iso_lines: Number of iso lines to compute
        :param parallel: Overwrite the default behaviour (parallel computation for 2 subshapes and more)
[... 137 lines cut ...]
            # Note: threejs uses point_normals: 
            #       https://threejsfundamentals.org/threejs/lessons/threejs-custom-buffergeometry.html
            
            # Extract relevant point_normals
            self._triangle_point_normals = np.asarray(self._poly_data.point_normals[self.triangle_offset:], dtype=np.float32)
        return self._triangle_point_normals
```

And we are down to 2.7 sec for my standard example:

```text
deflection= 0.95388, angular deflection= 0.30000, parallel=True
compute               : 2.2518911361694336
triangles             : 1.9073486328125e-06
triangle_points       : 0.00023484230041503906
triangle_point_normals: 0.3882560729980469
segments              : 0.055642127990722656

CPU times: user 15.1 s, sys: 521 ms, total: 15.6 s
Wall time: 2.7 s
```

[image]

@jmwright Since we discussed the C++ tessellator: I think, with OCP 7.5.1 we don't need it any more and should rely on the VTK tessellator.

jmwright, 2021-04-22:
@bernhard-42 

Have you committed your VTKTessellator anywhere yet?

bernhard-42, 2021-04-25:
@jmwright the current version is at https://github.com/bernhard-42/jupyter-cadquery/blob/ocp751/jupyter_cadquery/tessellator.py

bernhard-42, 2021-04-25:
@adam-urbanczyk 
Looks like `vtkPolyDataNormals` does not calculate the correct normal directions, see also https://tracker.dev.opencascade.org/view.php?id=25845. Hope this is solved in OCCT 7.6.0

The OCCT based result (all normals point outside):
[image]

The VTK based result (some normals point inside):
[image]

I still use the point normals, and not the face normals, but given the ticket above I do not expect any difference.

bernhard-42, 2021-04-25:
The wrong normal directions basically mean that I cannot us the VTK tessellator in *jupyter-cadquery* since *pythreejs* applies the shading according to the normals, which point into the wrong direction. For example, I get

[image]

instead of 

[image]

Playing around with `SetSplitting`, `SetConsistency`and `SetAutoOrientNormals` for the normal filter doesn't help either.

No idea how `plot_itk` solves that problem ...

So I think I have to stick with my OCCT tessellator until CadQuery has added OCCT 7.6.x

adam-urbanczyk, 2021-04-25:
Well, you do have the code of `plot_itk`. It should be fairly easy to find out what it does. My current result using VTK normals and VTK.js looks like this:
[image]

bernhard-42, 2021-04-25:
I don't fell anything is easy in *vtk* and *pyvista* world :-)

Basically the viewer seems to work totally different than *pythreejs*. I checked and the viewer of `itkwidgets` uses the same (wrong) normals. So the magic seems to happen somewhere else.

This just takes way too much time, so I will stick with my python version until the normals are fixed in VTK.

Anyhow, thanks for all your help. Really appreciated!

adam-urbanczyk, 2021-04-25:
Could be that the magic happens on the rendering (vtk.js) side then. Anyhows, the bug that you linked is resolved in master of OCCT, so indeed it should be solved in 7.6.

adam-urbanczyk, 2021-04-25:
For future generations: https://curvenote.com/@wesley/notes/!5G2LJYRxsBj6pKaQm1qo

TormodLandet, 2021-05-18:
I see that pull request https://github.com/CadQuery/cadquery/pull/735 has implemented this, but in case you (or someone else finding this in the future) want a solution that works without a dependency on the VTK Python package, the below code works with only vtk.js in the browser, no dependencies outside of cadquery (and IPython, but that is not required if you put the code inside a `_repr_javascript_` method).

I have not tested which solution produces the most compact JSON, embedded VTK files as in the PR or sending the data as lists as is done below. PS: the triangulator is a bit of a hack, but it just shows the concept.

Paste this into a jupyter lab cell to test:

```python
import json

from IPython.display import Javascript
import cadquery as cq
from cadquery.occ_impl.exporters import toCompound

def triangulate(assembly: cq.Assembly, tolerance:float=0.1, angularTolerance:float=0.1):
    """
    Trianulate an assembly into a list of dictionaries containing:
    - points: a flattened list of coordinates
    - polys: polygon node indices on vtk polydata format
      (num_nodes, node1, node2, node3, num_nodes, node1, node2 ....)
    - color: rgba in [0, 1]
    - name: part name
    """
    data = []
    
    def _triangulate(part):
        # Triangulate this assembly
        try:
            compound = toCompound(part.obj)
            vertices, indices = compound.tessellate(tolerance, angularTolerance)
            points = []
            for v in vertices:
                points.extend((v.x, v.y, v.z))
            polys = []
            for nodes in indices:
                polys.append(len(nodes))
                polys.extend(nodes)
            if part.color is not None:
                rgb = part.color.wrapped.GetRGB()
                alpha = part.color.wrapped.Alpha()
                color = (rgb.Red(), rgb.Green(), rgb.Blue(), alpha)
            else:
                color = (0.3, 0.9, 0.2, 1.0)
            data.append({
                "name": part.name,
                "color": color,
                "points": points,
                "polys": polys,
            })
        except AttributeError:
            pass

        # Triangulate any children
        for child in part.children:
            _triangulate(child)
    
    _triangulate(assembly)
    
    return data

JS_TEMPLATE = """
// ----------------------------------------------------------------------
function render_cadquery_model() {
    var actors = [];
    
    cadquery_data.forEach(function(item) {
        var polydata = vtk.Common.DataModel.vtkPolyData.newInstance();
        polydata.getPoints().setData(Float32Array.from(item.points), 3);
        polydata.getPolys().setData(Uint16Array.from(item.polys));
        
        var actor = vtk.Rendering.Core.vtkActor.newInstance();
[... 63 lines cut ...]
}
else {
    // Load the vtk.js library
    new Promise(function(resolve, reject) {
        var script = document.createElement("script");
        script.onload = resolve;
        script.onerror = reject;
        script.src = "https://unpkg.com/vtk.js@18.1.1/vtk.js";
        document.head.appendChild(script);
    }).then(() => {
        window.vtkjs_is_loaded = true;
        render_cadquery_model();
        //console.log("Loaded vtk.js library");
    });
}

"""

a = cq.Assembly()
a.add(cq.Workplane().box(2, 2, 1).faces('>Z').fillet(.5))
a.add(cq.Workplane().box(0.3, 0.3, 3), name='Red box', color=cq.Color(1.0, 0.3, 0.3, 0.8))
data = triangulate(a)

Javascript('var cadquery_data = ' + json.dumps(data) + ';\n\n\n\n' + JS_TEMPLATE)
```

TormodLandet, 2021-05-18:
The result should be a 3D view like below:
[image]

adam-urbanczyk, 2021-05-18:
Thanks for sharing! VTK is now anyhow a dep o OCP, so I don't think it is a problem. The embedded VTP uses data compression, so I expect it to be much smaller for large models. 

BTW: did you experience any issues regarding loading of vtk.js in jupyter? I cannot get it working in a stable way.

TormodLandet, 2021-05-18:
I need to rotate the model before using the scroll wheel in Firefox, otherwise the model disappears if I start by scrolling. Moving the camera slightly seems to work. You can move the camera somewhere, then call resetCamera() and it also still works. The code I posted seems stable for me, but a resetCamera should be added since the position is now fixed

adam-urbanczyk, 2021-05-18:
I was referring to vtk.js not being loaded properly. From your answer I get it is not an issue you've observed. I posted the question here: https://discourse.jupyter.org/t/trying-to-load-vtk-js-dynamically/9244

TormodLandet, 2021-05-18:
No, I have never observed that. My code above does ensure that the vtk.js code is loaded only once, but if you get errors on the first load then I do not know what the issue might be. Does it work after refreshing the Jupyter page in the browser? If you pass another function (to specify the onerror callback), does that get called?

TormodLandet, 2021-05-20:
If you write the output to a separate HTML document, bypassing Jupyter entirely, does it work then?

You can add the following code to my example above to generate a stand-alone test.html file:

```python
HTML_TEMPLATE = """
<!doctype html>
<html lang=en>
<head>
<meta charset=utf-8>
<title>Cadquery model</title>
</head>
<body>
<script>
window.onload = function main() {
var element = document.createElement('div');
document.body.appendChild(element);
JS_GOES_HERE
};
</script>
</body>
</html>
"""
js_string = 'var cadquery_data = ' + json.dumps(data) + ';\n\n\n\n' + JS_TEMPLATE
with open('test.html', 'wt') as f:
    f.write(HTML_TEMPLATE.replace('JS_GOES_HERE', js_string))
```

adam-urbanczyk, 2021-05-20:
Thanks, I know how to generate a standalone html ;). The issue is not with how to use vtk.js, but with how to load vtk.js. I think I've found a solution that works for me. I still do not understand why the promise approach does not work in jupyter (it does work in a standalone application). I asked about it here https://discourse.vtk.org/t/loading-vtk-js-dynamically/5755/3 and here https://discourse.jupyter.org/t/trying-to-load-vtk-js-dynamically/9244, but no luck so far.

bernhard-42, 2021-05-21:
@adam-urbanczyk 
The way you are adding javascript to Jupyter might only work for Jupyter Notebooks. AFAIK, JupyterLab will ignore this.

The correct way to add Javascript to Jupyter is to create a Jupyter Extension. It then renders in Jupyter Notebook, JupyterLab and even when you export the notebooks a s HTML.

I used `cookiecutter https://github.com/jupyter/widget-cookiecutter.git` to create such an extension and adapted your code into it, see https://github.com/bernhard-42/cadquery-vtk-viewer (commit 1 is the cookiecutter state, commit 2 the adaption to your code). The Readme might not work, however the script `dev-build.sh` does.
To test it you can just install it with `pip install git*https` from the repo (since JupyterLab 3 the Javascript stuff is hosten in python packages on pypi).

- JupyterLab
[image]

- Jupyter Notebook
[image]

TormodLandet, 2021-05-21:
I have similar code for another project (not open source unfortunately) that seems to work well in Lab without any extension. The code I pasted was also tested only in Jupyter Lab. It is a while since they re-enabled Javascript in Lab:
https://github.com/jupyterlab/jupyterlab/pull/4515
It might still be better to bundle an extension it that is the recommended way, but the `_repr_javascript_` method has been working for many years now, both in notebooks and lab I believe, I have not used classic notebooks for some time now

adam-urbanczyk, 2021-05-21:
Thanks @TormodLandet @bernhard-42, for now I only want to implement `_repr_javascript_` in #735 . If it is not working in some versions of JupyterLab, then too bad. It has been a yak shaving session anyway.

BTW: This seems to work reliably:
```javascript
new Promise(function(resolve, reject) {
        require.config({
        "paths": {"vtk": "https://unpkg.com/vtk"},
    }
);
    require(["vtk"], resolve);
}).then(() => {
    console.log(vtk);
    console.log("Loaded vtk.js library");
});
```

bernhard-42, 2021-05-21:
> It is a while since they re-enabled Javascript in Lab

nice, I wasn't aware, since I usually go for extensions. Especially, since JL3 they are really easy to maintain and install (before they were a pain in the neck)

The advantage of extensions are e.g. 
1) we could optimize the data transfer from python to javascript with `jupyterlab-datawidgets`. This is a proven way in projects like pythreejs.
2) No need to go for xml strings and parse them again on the Javascript side. One can simply transfer the numpy arrays. This should be better for larger modules.
3) Widgets allow to have realtime communication between the python and the javascript world. That means you can adapt your view later.

I am evaluating vtk.js as a replacement for pythreejs at the moment. So I needed the extension anyhow. If @adam-urbanczyk wants to go that route, happy to maintain, but he can also copy the code over to the cadquery org. It is really meant as a starting point, because at the beginning, extensions for widgets are pretty confusing ...

adam-urbanczyk, 2021-05-21:
The actual payloads in the xmls are gzipped binary data, so I don't think that writing of numpy arrays will be more efficient. I don't want to spend time on JupyterLab, but I invited you to the CQ org so that you can put the widget here (if you want to).

bernhard-42, 2021-05-21:
Cool, I will move the repo over then. 

The idea btw. is to use 

```python
from cadquery_vtk_viewer import VTKViewer
from IPython.display import display

class Workspace:
    # ...
    def _ipython_display_(self):
        data = {
            "shape": toString(self.val()),
            "color": [1, 0.8, 0, 1],
            "position": [0, 0, 0],
            "orientation": [0, 0, 0]
        }
        display(VTKViewer([data], width=400, height=300))
    # ...
```

Which would then expose the same user experience as the `_repr_javascript_` approach.

Btw. if you use the `pip install git+https` method, you need to have `npm` and `yarn` in the path which wouldn't be necessary when the package is published on pypi.

adam-urbanczyk, 2021-06-24:
Closed by #735

closed 2021-06-24 by adam-urbanczyk