CadQuery/cadquery #283: 'non-orthogonal GTrsf' Error When Calling multiply on a Matrix

jmwright, 2020-02-18:
The following code runs fine against master, but not against the OCP branch.
```python
from cadquery import Vector, Matrix

m = Matrix(
            [[-0.146655, -0.271161, -0.951296, 0.0376659],
            [-0.676234, 0.729359, -0.103649, 0.615421],
            [0.721942, 0.628098, -0.290333, -0.451955],
            [0, 0, 0, 1]]
    )
offset = Vector(0, 0, 0)
offset = m.multiply(offset)
```
When running it against the OCP branch, I get this:
```
>>> offset = m.multiply(offset)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 276, in multiply
    return other.transform(self)
  File "/home/jwright/Downloads/repos/cadquery/cadquery/occ_impl/geom.py", line 194, in transform
    pnt_t = pnt.Transformed(T.wrapped.Trsf())
OCP.Standard.Standard_ConstructionError: gp_GTrsf::Trsf() - non-orthogonal GTrsf
```
@adam-urbanczyk Any ideas?

adam-urbanczyk, 2020-02-18:
Nope, sounds like some internal change in OCCT.

jmwright, 2020-02-18:
I'm not sure what the original intent was, or why that specific matrix was used. It's from a [cqparts test](https://github.com/jmwright/cqparts/blob/master/tests/t_cqparts/test_coordsystem.py#L80). I believe that `cqparts.utils.CoodSystem` inherits from `cadquery.Plane`.

jmwright, 2020-02-18:
If the matrix is truly random as the comment right on line 73 suggests, maybe OCCT 7.4.0 is just catching an error that was ignored before?

adam-urbanczyk, 2020-02-19:
Indeed
 https://github.com/mmaenz/occt/blob/d42abf6789eb15032e1c4841f7887e8ee51971ea/src/gp/gp_GTrsf.lxx#L180
We might want to change this behavior actually.

adam-urbanczyk, 2020-02-20:
I was thinking - do we actually need to expose the matrix interface at all? There are nicer ways of setting the transformations.

jmwright, 2020-02-20:
Maybe not. I don't have a deep understanding of the use of the matrices in cqparts, but that functionality can probably be wrapped into the fluent API in CQ.