CadQuery/cadquery #285: How to set the color of a solid?

lalebarde, 2020-02-23:
It seems that it is possible to color solids as seen in [this thread](https://github.com/CadQuery/cadquery/issues/20), but I cannot find how to do this in the documentation. How to do it please?

adam-urbanczyk, 2020-02-23:
Currently it is not really supported. If you show multiple objects in CQ-editor you can set their color by hand. There is work to have this feature (again):
* https://github.com/CadQuery/cadquery/issues/276
* https://github.com/CadQuery/CQ-editor/pull/113
* https://github.com/cqparts/cqparts/issues/129

Note that only the first issue is "pure" CQ. What is your use case for color BTW?

lalebarde, 2020-02-24:
My use case is just to distinguish parts in an assembly in the 3d view.
The CQ-editor hand feature losses colors when you close the file.

adam-urbanczyk, 2020-04-02:
@lalebarde  https://github.com/CadQuery/CQ-editor/pull/127 has been merged and `show_object` implements all the color related options now.

lalebarde, 2020-04-16:
Works great, thanks!

```
show_object(result, name ='1')
show_object(result, name ='2', options=dict(alpha=0.5,color='red'))
show_object(result, name ='3', options=dict(alpha=0.5,color='#ff0000'))
show_object(result, name ='4', options=dict(alpha=0.5,color=(255,0,0)))
show_object(result, name ='5', options=dict(alpha=0.5,color=(1.,0,0)))
show_object(result, name ='6', options=dict(rgba=(1.,0,0,.5)))
show_object(result, name ='7', options=dict(color=('ff','cc','dd')))
```

closed 2020-04-16 by lalebarde

RagingRoosevelt, 2022-12-31:
How does this work if you want to export to, for example, 3mf?

jmwright, 2023-01-13:
> How does this work if you want to export to, for example, 3mf?

Colors are only preserved for assemblies on STEP export. STL, 3MF, etc lose color info when exported.