CadQuery/cadquery #286: Change Deprecated time.clock() Call in CQGI

jmwright, 2020-02-24:
A user reported this on the cadquery-freecad-module project, but it applies to CQ 2.0 as well.

https://github.com/jmwright/cadquery-freecad-module/issues/147

The gist of it is that `time.clock()` has been deprecated since Python 3.3, and was removed in Python 3.8. We need to switch the CQGI references over to either `time.perf_counter()` or `time.process_time()`.

Reference 1: https://github.com/CadQuery/cadquery/blob/102c16c14e42b6acca484efbf017f9ac2cc75789/cadquery/cqgi.py#L99

Reference 2: https://github.com/CadQuery/cadquery/blob/102c16c14e42b6acca484efbf017f9ac2cc75789/cadquery/cqgi.py#L124

jwhevans, 2020-03-22:
Do you have a preference? time.perf_counter() is generally more accurate.

jmwright, 2020-03-22:
I think `perf_counter` is fine. If I understand correctly, the main difference is it includes sleep time and interrupts, and `process_time` doesn't.

jmwright, 2020-03-28:
Fixed in #305

closed 2020-03-28 by jmwright