CadQuery/cadquery #287: conda-forge package

looooo, 2020-02-25:
Now that pythonocc is packaged via conda-forge I wanted to ask if it is likely to see cadquery packaged for conda-forge too.

- As far as I know, previously cadquery depend on freecad and now changed to pythonocc. Is this correct?
- Is there a release planned? When can we expect a release?

I would like to keep growing the cad-packages on conda-forge and try to keep the occt-dependent packages compatible.

adam-urbanczyk, 2020-02-25:
Thanks for reaching out @looooo ! CQ and CQ-editor have conda packages in the CadQuery organization. Indeed it would be good to have on conda-forge. To answer your specific questions:

* Currently CQ is based on PythonOCC
* We are close to the 2.0 relese.

Currently CQ uses not the latest PythonOCC and a custom for of it (with modifications related to CQ-editor display). We are still investigating how to move to OCCT7.4. Thanks a lot BTW for maintaining that conda package.

bollwyvl, 2021-02-09:
Hey, folks! Just saw the PR for 1.2.5 EOL release: https://github.com/conda-forge/cadquery-feedstock/pull/3

Since last we jawed, things have gotten nicer for experimental/performance stuff over on CF:
- support for rc releases on dedicated channels
- automerge (to be used judiciously, for well-tested packages) 
- ppy38, 39, pypy (plus all the fixin's like scipy, numpy, etc)
- osx ARM (through some bizarre, heroic feats of transcompilation)
- miniforge/mambaforge for a fully community-curated, ToS-unencumbered toolchain, even on winders
- the setup-miniconda (plus all those other installers) github actions

meanwhile, on the jupyter stuff: JupyterLab 3 means otherwise magical technology doesn't get overshadowed by nodejs build problems (zzzz).

Anyhow, happy to help get the 2.x line onto conda-forge!

shimwell, 2022-02-25:
I wonder if it might be worth having another look at this issue as things have moved on a bit in the last two years. I'm wondering if something like this might work

```

{% set name = "cadquery" %}
{% set version = "2.1" %}

package:
  name: {{ name|lower }}
  version: {{ version }}

source:
  url: https://github.com/CadQuery/cadquery/archive/refs/tags/2.1.tar.gz
  sha256: 574e3674a6f97fb4ae3849e7453bbb341e14c8675642aa1887b30b177b5c7133

build:
  number: 0
  script: "{{ PYTHON }} -m pip install . -vv"

requirements:
  build:
    - python
    - setuptools
  host:
    - python {{ PYTHON }}
    - pip
  run:
    - python {{ PYTHON }}
    - ocp = 7.5.3
    - pyparsing >= 2.1.9
    - ezdxf
    - ipython
    - typing_extensions
    - nptyping
    - nlopt
    - multimethod = 1.6

test:
  source_files:
    - tests/
  imports:
    - cadquery
  requires:
    - pytest
    - docutils
    - path
  commands:
    - pytest -v

about:
  home: https://github.com/CadQuery/cadquery
  license: Apache
  license_family: Apache
  license_file: LICENSE
  summary: 'A python parametric CAD scripting framework based on OCCT '
  # The remaining entries in this section are optional, but recommended.
  description: |
    CadQuery is an intuitive, easy-to-use Python module for building parametric
    3D CAD models. Using CadQuery, you can write short, simple scripts that
    produce high quality CAD models. It is easy to make many different objects
    using a single script that can be customized.
  doc_url: https://cadquery.readthedocs.io/
  dev_url: https://github.com/CadQuery/cadquery

extra:
  recipe-maintainers:
    - adam-urbanczyk
    - jmwright
    - marcus7070
    - dcowden
```