CadQuery/cadquery #288: rotate

KeithSloan, 2020-02-26:
[https://pythonhosted.org/cadquery/classreference.html#cadquery.Workplane](url)

Gives the parameters of rotate as
rotate(start, end, degrees)

But only mentions start (3 tuple of floats) and degrees (float)

Also not sure how the 3 floats specify an axis of rotation

Could you please supply an example.

Thanks

jmwright, 2020-02-26:
The pythonhosted link doesn't work for me. Are you looking for something like this?

https://github.com/CadQuery/cadquery/blob/master/examples/Ex015_Rotated_Workplanes.py

KeithSloan, 2020-02-26:
Not sure what went wrong with me trying to post link. Try this one
https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.CQ.rotate

adam-urbanczyk, 2020-03-02:
Thanks for reporting, the docs need fixing. NB: to specify an axis you need two points (Arg1 and Arg2)