CadQuery/cadquery #290: Create wire from selected vertices

r4dr3fr4d, 2020-03-01:
I select vertices from different solids (e.g., corners of different boxes), and I want to turn four of them into a wire that I can loft to another wire made in the same fashion.  Is there a way to just create a wire (or maybe an edge or face) given some 3d points, without involving a workplane?

marcus7070, 2020-03-01:
This was a bit awkward, but here is one way to do it.

```python
import cadquery as cq

boxes = (
    cq
    .Workplane("XY")
    .transformed(rotate=(0, 0, 45))
    .rect(5, 5, forConstruction=True)
    .vertices()
    .box(1, 1, 1)
)

# create a list of 3d points from selected vertices
pnts = []
for sel in [">X", ">Y", "<X", "<Y"]:
    vertex = boxes.faces(">Z").vertices(sel).val()
    # cq.Wire.makePolygon requires Vectors, so convert Vertices to Vectors
    pnts.append(cq.Vector(vertex.toTuple()))
pnts.append(pnts[0])  # close the wire
wire0 = cq.Wire.makePolygon(pnts)

loft0 = (
    cq
    .Workplane("XY", origin=(0, 0, 5))
    .circle(1)
)
# methods like loft find the wires to extrude from in CQ.ctx.pendingWires,
# so we have to add wire0 to pendingWires
loft0._addPendingWire(wire0)
loft0 = loft0.loft()
```
[image]

r4dr3fr4d, 2020-03-02:
Looks like makePolygon() was what I was looking for!!!  Thanks so much @marcus7070!

closed 2020-03-02 by r4dr3fr4d