CadQuery/cadquery #291: Can I return a Python OCC Shape for cadquery to display,

KeithSloan, 2020-03-02:
For example
def getShape(self) :
          from OCC import BRepPrim
          print("Cone : getShape")
          return(OCC.Core.BRepPrim.BRepPrim_Cone(self.R1[1], \
                 self.R2[1], self.Z))

And the main program does show_object(shape).

KeithSloan, 2020-03-02:
Okay I see that the OCC.Core does not actually return a shape. So question has is cadquery creating shapes with OCC Python to display.

adam-urbanczyk, 2020-03-02:
There is no `show_object` in CQ. You are probably referring to CQ-editor. For pure PythonOCC object to work you'd need to wrap it into `CQ.Shape` first

KeithSloan, 2020-03-03:
Must be missing something.

Without calling my module I tried in cq-editor script

`from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
ret = (BRepPrimAPI_MakeBox(100., 100., 100.).Shape()
show_object(ret)`

and

`from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
ret = cq.CQ(BRepPrimAPI_MakeBox(100., 100., 100.).Shape())
show_object(ret)`

Neither of which display any box

jmwright, 2020-03-03:
@KeithSloan Using the PythonOCC API directly, this works for me in CQ-editor:
```python
from cadquery import Shape
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

ret = BRepPrimAPI_MakeBox(gp_Ax2(gp_Pnt(-0.1, -1.0, -1.5), gp_Dir(0, 0, 1)), 1.0, 2.0, 3.0).Shape()

# show_object needs a CadQuery object
result = Shape.cast(ret)

show_object(result)
```
Your code snippets above throw errors for me and so they don't display anything. There is [this reference](https://github.com/CadQuery/cadquery/blob/e41606975967e7270c642ac1d5ad38a9f4eac89f/doc/extending.rst#using-pythonocc-script) in the docs, but I can't get the mixing in of `Vector` like that to work. I needed to use `gp_Pnt` and `gp_Dir` instead.

closed 2020-05-05 by jmwright