CadQuery/cadquery #292: Workplane.spline() with tangents is not planar

goatchurchprime, 2020-03-04:
I was trying to make a lofting function between a series of splined shapes when I discovered that when you attempt to set the endpoint tangents on a spline shape in a workplane that is offset from the origin, the shape is distorted out of the plane.  

This happens for all workplane orientations (XZ and YZ) if you adjust the workplane origin to take it away from the plane through the origin, so it's a consistent bug, not just a typo in one of the conditions.  

```Python
import cadquery as cq
pts = [(0,0), (-1,1), (-2,0), (-1,0)]
tangents = [(0,1), (1,0)]

print("\nWith tangents applied")
s = cq.Workplane("XY", (0,0,1))
result = s.spline(pts, tangents=tangents, makeWire=True)
bb = result.objects[0].BoundingBox()
print("xrange (%f,%f)" % (bb.xmin, bb.xmax))
print("yrange (%f,%f)" % (bb.ymin, bb.ymax))
print("zrange (%f,%f)" % (bb.zmin, bb.zmax))
     
print("\nTangents=None")
s = cq.Workplane("XY", (0,0,1))
result = s.spline(pts, tangents=None, makeWire=True)
bb = result.objects[0].BoundingBox()
print("xrange (%f,%f)" % (bb.xmin, bb.xmax))
print("yrange (%f,%f)" % (bb.ymin, bb.ymax))
print("zrange (%f,%f)" % (bb.zmin, bb.zmax))
```

>With tangents applied
>xrange (-2.020449,0.010000)
>yrange (-0.069845,1.012532)
>zrange (0.724149,1.246722)
>
>Tangents=None
>xrange (-2.009654,0.010000)
>yrange (-0.183280,1.084804)
>zrange (0.990000,1.010000)

(If you do use a workplane through the origin the spline is planar, even with tangents.)

adam-urbanczyk, 2020-03-04:
Confirmed bug. Thanks for reporting.

adam-urbanczyk, 2020-03-06:
I should be fixed by #293 @goatchurchprime

PR #293 (2020-03-06): Spline tangnet fix for offset workplane
Will resolve #292
commit b9d3a1c1 (2020-03-06): Spline tangnet fix for offset workplane (#293)

Fixes #292
closed 2020-03-06 by adam-urbanczyk