CadQuery/cadquery #295: selection given an object's local coordinate system?

r4dr3fr4d, 2020-03-08:
Is it possible to use string selection to select components of a solid that has been translated/rotated?  For example, if you have a box that has been rotated 45 degrees (meaning part of the top is in the negative Z space), still use s('>Z') to get the box's "top" face or verticies?

adam-urbanczyk, 2020-03-08:
You can use vectors in the selectors, e.g. `... .faces('>(0,1,1)')`. Is that what you mean?

r4dr3fr4d, 2020-03-09:
Hmmm don't think so.
```
test_vector1 = cq.Workplane("front").box(switch_width, switch_length, plate_thickness).faces('>(0,0,1)')
test_vector2 = cq.Workplane("front").box(switch_width, switch_length, plate_thickness).rotate((0,0,0), (0,1,0), 45).faces('>(0,0,1)')
show_object(test_vector1)
show_object(test_vector2)
```

These result in two different faces being selected since the selection vector is relative to the origin.  Can we make it relative to the box itself?  Does that make sense?

Thanks for the reply btw.

adam-urbanczyk, 2020-03-09:
I understand what you mean. Selectors use absolute coordinates. I was just pointing to the fact that you can use any vector. You should be able to transform from local to global using `... .plane.toWorldCoord(v)`

r4dr3fr4d, 2020-03-18:
I messed with it some and looked at other examples like #280.  I would think that `test_vect = cq.Vector(test_vector2.plane.toWorldCoords((0,0,1)))` would yield a new vector with numbers other than (0,0,1), but that's all I get so far.  Even after that, I'm not sure how to feed that into the selector, but I'm not there yet.

adam-urbanczyk, 2020-03-19:
Once you have the vector say `v`, you could do something like:
```python
... .faces(f'>{v.toTuple()}')
```
or
```
... .faces(cq.PerpendicularDirSelector(v))
```
If you paste a more complete example I can be more specific.

r4dr3fr4d, 2020-03-22:
```
test_vector1 = cq.Workplane("front").box(switch_width, switch_length, plate_thickness).faces('>(0,0,1)')
test_vector2 = cq.Workplane("front").box(switch_width, switch_length, plate_thickness).rotate((0,0,0), (0,1,0), 45).faces('>(0,0,1)')
v = cq.Vector(test_vector2.plane.toWorldCoords((0,0,1)))
test_vector2 = test_vector2.faces(cq.PerpendicularDirSelector(v))
show_object(test_vector1)
show_object(test_vector2)
```
If I select via the format string, it throws a ParseException, I'm afraid.  My goal is for the two objects (test_vectors 1 and 2) to show the same face, given this code.

r4dr3fr4d, 2020-03-22:
I would think that 'plane' in line 3 would be coordinates local to the rotated box?  Am I mistaken?

r4dr3fr4d, 2020-05-28:
Update - still no good, below is another example - as you edit the angle in rotate, the selected/displayed vertices on the box will change, when the desired behavior is that the selection should never change regardless of the object's previous transformations.

'''
est = cq.Workplane("front")
#tests = cq.Workplane("front").box(10,10,10).rotate((0,0,0), (1,0,0), 0)
#tests = cq.Workplane("front").box(10,10,10).rotate((0,0,0), (1,0,0), -80)
tests = cq.Workplane("front").box(10,10,10).rotate((0,0,0), (1,0,0), -40)
v = test.plane.toWorldCoords((0,1,1))
#v = test.workplane().plane.toWorldCoords((0,1,1))
verts = tests.vertices(cq.DirectionMinMaxSelector(v))

show_object(verts)
show_object(tests)
'''

r4dr3fr4d, 2020-06-13:
Messed with this some more this evening and I think NearestToPointSelector(<point_relative_to_plane_origin>) works best for me.  I could never get the directional ones working and not sure how I can since they seem to reference the given direction from the world origin.  I will close, pending any other comments.

closed 2020-06-13 by r4dr3fr4d