CadQuery/cadquery #296: How to buld a 3D spline in cadquery?

lalebarde, 2020-03-10:
I need to build objects in 3D by sweeping a shape over a 3D-spline. After reviwing the available commands, it looks that splines are only available in 2D in cadquery, is it true? If not, how could I do it with OpenOCC please?

marcus7070, 2020-03-10:
https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Edge.makeSpline will make a spline from a list of Vectors (ie. 3 dimensional). You need to wrap the resulting edge in a CQ object to use it with sweep. I'll try to write an example now, I'm thinking this might be a good addition to the docs in order to introduce users to the occ_impl layer.

marcus7070, 2020-03-10:
```python
import cadquery as cq
from random import random

list_of_vecs = [cq.Vector(random(),random(), i + random()) for i in range(-5, 5, 2)]
spline_edge = cq.Edge.makeSpline(list_of_vecs)
spline_cq = cq.Workplane("XY")
spline_cq.objects = [spline_edge]
part = (
    cq
    .Workplane("XY", origin=list_of_vecs[0])
    .circle(1)
    .sweep(spline_cq)
)
```
[image]

That `spline_cq.objects = [spline_edge]` is an ugly workaround because `cq.Workplane.sweep` calls `path.wire()`, so the path needs to be a `cq.Workplane` object. I hope someone else can tell me a better way to go from an Edge object to a path suitable for sweep.

marcus7070, 2020-03-10:
Replacing 
```python
spline_cq = cq.Workplane("XY")
spline_cq.objects = [spline_edge]
```

with 
```python
spline_cq = cq.Workplane("XY").newObject([spline_edge])
```
is an improvement I think.

jmwright, 2020-03-11:
I think that `newObject` is the accepted way to do it. That's the way it's done internally in the codebase.

adam-urbanczyk, 2020-03-11:
You can actually just use 3D coordinates like so:
```python
pts = [
    (1,1,1),
    (1,2,2),
    (1,3,3),
]
p= cq.Workplane("XY").spline(pts,includeCurrent=True)
s = cq.Workplane("XY").rect(1,1).sweep(p)
```

[image]

adam-urbanczyk, 2020-03-11:
This looks actually like a docs issue, do you agree @jmwright  @dcowden ?

jmwright, 2020-03-11:
It looks that way to me.

lalebarde, 2020-04-14:
Great, thanks a lot all of you!

lalebarde, 2020-04-14:
In fact, a cadquery Workplane is defining a 3D geometric frame of reference isn't it?

lalebarde, 2020-04-16:
Yes and it is [in the documentation!](https://cadquery.readthedocs.io/en/latest/primer.html#workplanes) Blame on me.

closed 2020-04-16 by lalebarde