CadQuery/cadquery #297: Allow sweep path argument to be a wire

marcus7070, 2020-03-10:
Currently `Workplane.sweep` calls `path.wire()`, so path has to be another `Workplane` object.

https://github.com/CadQuery/cadquery/blob/b9d3a1c18704edba7361256622acd767456240f3/cadquery/cq.py#L2466-L2512

But if someone was to construct a spline directly, like #296, then they have to turn it into a Workplane object just for `sweep` to turn it back into a Wire. 

Perhaps `sweep` could be changed to handle either an Edge, Wire or Workplane object as the path argument?

marcus7070, 2020-03-10:
Note that `Workplane.sweep` winds up calling `occ_impl.Solid.sweep` or `sweep_mutli` and both do type checking to handle Edge objects.

jmwright, 2020-03-11:
I would be fine with this change.

adam-urbanczyk, 2020-03-11:
As I mentioned in #296 one can just use full coordinates to construct splines - there is no need to call the `occ_impl` layer. In view of this fact can we close this issue to not complicate the API? BTW: the underlying assumption is that a regular user will just call the high-level API.

marcus7070, 2020-03-11:
OK @adam-urbanczyk, but is `Workplane.spline` the supported method of making 3D splines? The docs specifically state "list of 2-tuples", and none of our tests cover making a spline in 3D. Also I think that would make spline the only Workplane method that returns a 3D edge/wire? 

Regarding complicating the API, this is what I am suggesting:
```diff
diff --git a/cadquery/cq.py b/cadquery/cq.py
index 65e628b..5e081d6 100644
--- a/cadquery/cq.py
+++ b/cadquery/cq.py
@@ -2551,9 +2551,11 @@ class Workplane(CQ):
                 "be removed in the next version; use multisection instead",
                 DeprecationWarning,
             )
+        if isinstance(path, Workplane):
+            path = path.wire()
 
         r = self._sweep(
-            path.wire(), multisection, makeSolid, isFrenet, transition
+            path, multisection, makeSolid, isFrenet, transition
         )  # returns a Solid (or a compound if there were multiple)
         if combine:
             newS = self._combineWithBase(r)
```

Also a new test and some changes to the docs that I don't have the time to write right now.

adam-urbanczyk, 2020-03-11:
I don't see why the spline method should only work for 2D. Of course the docs need to be updated. If that is in place do you still see the need for adding this functionality?

marcus7070, 2020-03-11:
No need then.

At first I found it unintuitive to create a 3D path from the conceptually 2D workplane, but after thinking about it for a bit it actually seems quite useful. 

eg. Create a workplane anywhere and with any orientation, then create a 2D section to be swept. Create a new CQ object for the path, use `copyWorkplane(obj)` to get the workplane of the section, then your spline can start at coords (0, 0, 0) and you can specify your spline (and tangents) in workplane coords. Much easier than converting spline points to global coords.

closed 2020-03-11 by marcus7070