CadQuery/cadquery #302: Virtual env

zignig, 2020-03-24:
Now that we have a  binary artifact server  (https://github.com/CadQuery/OCP) it should be possible to create a Python Virtual Env as a script.

I like and use conda , but I think that there is a need to have a non conda install path available.

I am looking for some direction in creating a script to build a working environment.

@adam-urbanczyk  @jmwright ?

This is with a view to get cqparts into CQ2 and then possibly integration into CQ-editor.

Good work all, this is going to be a welcome distraction as I have suddenly got some free time.

adam-urbanczyk, 2020-03-24:
You could try using [conda-press](https://github.com/regro/conda-press) for converting conda packages into wheels (see #153). The problem with using pip is that it does not really support non-python dependencies.

Please leave OCP out of this discussion for now. I did release the binaries just to easy testing of OCP. I will sooner or later build a proper CI pipeline for it.

zignig, 2020-03-24:
@adam-urbanczyk , I think you misunderstand my intent.

I want to script up a bleeding edge cadquery builder for a fork of cqparts i am restarting work on.

I'll try out the https://github.com/CadQuery/cadquery/tree/OCP branch and work from there.

:)

zignig, 2020-03-24:
Sorry, that may have appeared rude.

can you please give some hand_wavy directions to fiddle with a OCP + cqparts -  !freecad build ? 

please.

jmwright, 2020-03-24:
@zignig On Linux while I was updating cqparts for CQ 2.0, I just placed the OCP library in the cadquery directory and then set PYTHONPATH to include the path to the CQ directory. Then I could run the cqparts tests on my fork. Is that the kind of directions you're looking for?

adam-urbanczyk, 2020-03-24:
No worries @zignig . I indeed thought you meant something else. So for now you'll need:

- OCCT 7.4 (I get it from the conda-forge channel)
- OCP branch of CQ + the python deps of CQ
- OCP binary on your PYTHONPATH (note the builds are for py3.6)

zignig, 2020-03-24:
Excellent , thank you both.

zignig, 2020-03-28:
Cool ! , it works. 

The gltf export is broken ( the winding / point order is borked ). But the cqparts stuff works with OCP.

jmwright, 2020-03-28:
@zignig Great news! Can you post a link to your project?

`transposed list` in the gltf codec is only there to support cqparts I think, and is not used anywhere else as far as I know. It could be broken and we wouldn't know it. Scroll down to the gltf.py change here: https://github.com/jmwright/cqparts/commit/c41cfabca36f30f73290bd02a634e539e2a36031#

I also updated how the matrix worked here to update it. https://github.com/jmwright/cqparts/commit/a2517631a877b20515e529cc17f440e71b6bab14

There are other places gltf could have gotten broken, but those things I listed above are specific CQ functionality for cqparts and are related to gltf export (at least I think the second one is).

I don't have an overly deep understanding of the cqparts codebase, so any help figuring out why gltf export is broken would be appreciated. Maybe at least an example showing what's broken would be helpful.

zignig, 2020-03-29:
I will build a clean ubuntu18 virtual machine so the dependencies are covered.  The repo will be in 

https://github.com/zignig/OCT_cqparts  once I have it in a usable state.

zignig, 2020-03-29:
A 10x10x10 unit cube renders like this as a GLTF

https://zignig.github.io/research/index.html

An STL export of the same construct works fine , so it is probably the cqparts gltf export code.

zignig, 2020-07-10:
closing , waiting for CQ2.0

closed 2020-07-10 by zignig