CadQuery/cadquery #303: newObject and cut() method

bragostin, 2020-03-25:
I have created a newObject and want to perform a cut like this:
```python
cut_new = cq.Workplane('YZ').box(L,W,H)
new = cq.Workplane('XY').newObject(list_of_shapes)
cut_new = new.cut(shape_to_cut)
```
but nothing happens: the ``new`` object remains unchanged. Is that to be expected?

adam-urbanczyk, 2020-03-25:
I don't think this is correct. Could you share a minimum reproducible example?

bragostin, 2020-03-25:
@adam-urbanczyk 
This produces the intended result:
```python
import cadquery as cq
box1 = cq.Workplane('XY').box(10, 20, 30)
box2 = cq.Workplane('YZ').box(10, 20, 30)
shape_to_cut = cq.Workplane('XY').box(15, 15, 15).translate((8,8,8))
new_object = box1.union(box2)
cut_object = new_object.cut(shape_to_cut)
```
[image]
while this does not:
```python
import cadquery as cq
box1 = cq.Workplane('XY').box(10, 20, 30)
box2 = cq.Workplane('YZ').box(10, 20, 30)
shape_to_cut = cq.Workplane('XY').box(15, 15, 15).translate((8,8,8))
list_of_shapes = []
for o in box1.all(): list_of_shapes.extend(o.vals())
for o in box2.all(): list_of_shapes.extend(o.vals())
new_object = cq.Workplane('XY').newObject(list_of_shapes) 
cut_object = new_object.cut(shape_to_cut)
```
[image]
I would like to avoid using ``union`` because it is exceedingly slow with my real model.

adam-urbanczyk, 2020-03-25:
Confirmed, it might be a kernel issue.

adam-urbanczyk, 2020-03-27:
I made some progress: it seems that OCCT bool ops do not like if the original shape has self intersections. In your example the two boxes d intersect. For now the workaround would be to perform the cut before constructing the compound.

I will still try if using this method allows to pefrom such an op in a single shot
https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_b_rep_algo_a_p_i___builder_algo.html#a46ae85feda209da3074b8b77af6670a7

bragostin, 2020-03-28:
@adam-urbanczyk  then maybe a workaround could be to apply the the cut on each object of the list separately in a for loop?

adam-urbanczyk, 2020-03-28:
Indeed, cut on separate objects and only then construct the compound.

bragostin, 2020-03-28:
Indeed, the following code gives the expected result without unions:
```python
import cadquery as cq

box1 = cq.Workplane('XY').box(10, 20, 30)
box2 = cq.Workplane('YZ').box(10, 20, 30)
shape_to_cut = cq.Workplane('XY').box(15, 15, 15).translate((8,8,8))
list_of_shapes = []
for o in box1.all(): list_of_shapes.extend(o.vals())
for o in box2.all(): list_of_shapes.extend(o.vals())
list_of_shapes = [ cq.Workplane('XY').newObject([o]).cut(shape_to_cut) for o in list_of_shapes ]
list_of_shapes = [ o.val() for o in list_of_shapes ]
cut_object = cq.Workplane('XY').newObject(list_of_shapes)
```

PR #312 (2020-04-07): Bool op refactoring
Will resolve #303

- [x] Add `__iter__` to `Compound`
- [x] Refactor bool ops in `shapes.py`
- [x] Refactor bool ops in `cq.py`
- [x] Add tests

Additionally, this PR speeds up bool operation for some cases, e.g. repetitive patterns. It should at least partially solve  #183 as well.
closed 2020-04-07 by jmwright