CadQuery/cadquery #304: Rotations seem to be lost when using loft

r4dr3fr4d, 2020-03-27:
```s = cq.selectors.StringSyntaxSelector

box = cq.Workplane("front")
box = box.box(10, 10, 3)

def transform(s, r, c):
    return s.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# example to flush the left face
test =  box.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# the list of points with which we make the wire need to be in rotational order
left1 = transform(box.vertices(s("<X and >Y and >Z")), 0, 0).val()
left2 = transform(box.vertices(s("<X and >Y and <Z")), 0, 0).val()
left3 = transform(box.vertices(s("<X and <Y and <Z")), 0, 0).val()
left4 = transform(box.vertices(s("<X and <Y and >Z")), 0, 0).val()
left_ordered = [left1, left2, left3, left4, left1]
tups = [v.toTuple() for v in left_ordered]

left_face = cq.Wire.makePolygon([cq.Vector(t) for t in tups])

# get minimum value for flush and rewrite the vectors
min_tup = min([t[0] for t in tups])
left_flush = [(min_tup, t[1], t[2]) for t in tups]
vects = [cq.Vector(t) for t in left_flush]
left_flush = cq.Wire.makePolygon(vects)

# here's where the issue lies
# uncomment these lines, and the box seems to lose the second rotation in transform(), and no longer aligned with the left_flush solid
#test.ctx.pendingWires = [left_face, left_flush]
#test = test.loft(combine=True)

show_object(test)
show_object(left_face)
show_object(left_flush)
```
Not sure if this is due to me misunderstanding how the stack works.  The lines with `test.ctx.pendingWires` and `test.loft()` seem to mess with the rotation of test, by removing the most recent operation (see the two rotate's in transform()).

With those two lines commented out, the box is of the desired rotation and lines up with the wires to be lofted:
[image]
But when I uncomment the lines to complete the loft, the second/last rotation seems to be lost:
[image]

adam-urbanczyk, 2020-03-28:
Could you post a minimum working example? In the code above `s` seems to be undefined.

r4dr3fr4d, 2020-03-28:
Updated :)  Apparently the triple tick marks need to be on their own line.

r4dr3fr4d, 2020-04-11:
So I have a bit of a workaround:  if I set `combine=False`, and set the loft result to a different name, I can use that solid and the solid prior to lofting together to get the proper result:

```
s = cq.selectors.StringSyntaxSelector

box = cq.Workplane("front")
box = box.box(10, 10, 3)

def transform(s, r, c):
    return s.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# example to flush the left face
test =  box.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# the list of points with which we make the wire need to be in rotational order
left1 = transform(box.vertices(s("<X and >Y and >Z")), 0, 0).val()
left2 = transform(box.vertices(s("<X and >Y and <Z")), 0, 0).val()
left3 = transform(box.vertices(s("<X and <Y and <Z")), 0, 0).val()
left4 = transform(box.vertices(s("<X and <Y and >Z")), 0, 0).val()
left_ordered = [left1, left2, left3, left4, left1]
tups = [v.toTuple() for v in left_ordered]

left_face = cq.Wire.makePolygon([cq.Vector(t) for t in tups])

# get minimum value for flush and rewrite the vectors
min_tup = min([t[0] for t in tups])
left_flush = [(min_tup, t[1], t[2]) for t in tups]
vects = [cq.Vector(t) for t in left_flush]
left_flush = cq.Wire.makePolygon(vects)

# here's where the issue lies
# uncomment these lines, and the box seems to lose the second rotation in transform(), and no longer aligned with the left_flush solid
test.ctx.pendingWires = [left_face, left_flush]
test_loft = test.loft(combine=False)

show_object(test)
show_object(test_loft)
#show_object(left_face)
#show_object(left_flush)
```

[image]

That's what I _want_ to see with the original example. 

Definitely not ideal, but I can work with it.  I've looked at this for a bit now, and still no ideas as to why loft() seems to be messing with the stack.

adam-urbanczyk, 2020-04-11:
This almost works for me (not the additional `workplane()` call before lofting):
```python
s = cq.selectors.StringSyntaxSelector

box = cq.Workplane("front")
box = box.box(10, 10, 3)

def transform(s, r, c):
    return s.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# example to flush the left face
test =  box.rotate((0,0,1), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# the list of points with which we make the wire need to be in rotational order
left1 = transform(box.vertices(s("<X and >Y and >Z")), 0, 0).val()
left2 = transform(box.vertices(s("<X and >Y and <Z")), 0, 0).val()
left3 = transform(box.vertices(s("<X and <Y and <Z")), 0, 0).val()
left4 = transform(box.vertices(s("<X and <Y and >Z")), 0, 0).val()
left_ordered = [left1, left2, left3, left4, left1]
tups = [v.toTuple() for v in left_ordered]

left_face = cq.Wire.makePolygon([cq.Vector(t) for t in tups])

# get minimum value for flush and rewrite the vectors
min_tup = min([t[0] for t in tups])
left_flush = [(min_tup, t[1], t[2]) for t in tups]
vects = [cq.Vector(t) for t in left_flush]
left_flush = cq.Wire.makePolygon(vects)

# here's where the issue lies
# uncomment these lines, and the box seems to lose the second rotation in transform(), and no longer aligned with the left_flush solid
test.ctx.pendingWires = [left_face, left_flush]
test = test.workplane().loft(combine=Ture)

show_object(test)
show_object(left_face)
show_object(left_flush)
```
It seems that the resulting solids is invalid though. It is some kind of kernel glitch. TBC...

r4dr3fr4d, 2020-06-09:
I found another issue with this one, not sure if related:
```
s = cq.selectors.StringSyntaxSelector

box = cq.Workplane("front")
box = box.box(10, 10, 3)

def transform(s, r, c):
    return s.rotate((0,1,0), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# example to flush the left face
test =  box.rotate((0,1,0), (0,0,0), 10).rotate((0,1,0), (0,0,0), 10)

# the list of points with which we make the wire need to be in rotational order
left1 = transform(box.vertices(s("<X and >Y and >Z")), 0, 0).val()
left2 = transform(box.vertices(s("<X and >Y and <Z")), 0, 0).val()
left3 = transform(box.vertices(s("<X and <Y and <Z")), 0, 0).val()
left4 = transform(box.vertices(s("<X and <Y and >Z")), 0, 0).val()
left_ordered = [left1, left2, left3, left4, left1]
tups = [v.toTuple() for v in left_ordered]

left_face = cq.Wire.makePolygon([cq.Vector(t) for t in tups])

# get minimum value for flush and rewrite the vectors
extend = .1
min_tup = min([t[0] for t in tups]) - extend
left_flush = [(min_tup, t[1], t[2]) for t in tups]
vects = [cq.Vector(t) for t in left_flush]
left_flush = cq.Wire.makePolygon(vects)

# here's where the issue lies
# uncomment these lines, and the box seems to lose the second rotation in transform(), and no longer aligned with the left_flush solid
test.ctx.pendingWires = [left_face, left_flush]
test = test.workplane().loft(combine=True)

show_object(test)
show_object(left_face)
show_object(left_flush)
```
Note the `extend` variable.  If you set that to 0, it'll throw a 'command not done' error.  I wonder if it has something to do with the loft wires sharing vertices, in this case two of them?

adam-urbanczyk, 2022-09-23:
Is your `test` valid (`test.val().isValid()`) @lorenzncode ? I do get false, but `fix()` works in this case.

lorenzncode, 2022-09-23:
No, not valid without ```fix()```.  I also get False for the ```isValid()``` check.  I didn't notice that before.