CadQuery/cadquery #306: CQGI debug use case not clear

adam-urbanczyk, 2020-04-02:
I do not get what `debug` is bringing on top of `show_object` and how would I implement it. Shall we maybe remove it completely from the API @jmwright @dcowden ?

jmwright, 2020-04-02:
In the FreeCAD workbench, I would display debug objects in red with transparency.

https://github.com/jmwright/cadquery-freecad-module/blob/master/CQGui/Command.py#L213

I think it was designed as a way for people to show intermediate steps in the build process, but I don't know if anybody ever used it. With the ability to step through a model with CQ-editor, maybe it's not needed. I have heard one or two users mention debug objects, but I have no idea what percentage make use of that feature.

tolot27, 2020-04-02:
I find visual debugging as it is implemented in [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) quite useful.

adam-urbanczyk, 2020-04-04:
@tolot27 thanks, but that was not really the question. `debug()` is in the CQGI specification: https://cadquery.readthedocs.io/en/latest/cqgi.html#cadquery.cqgi.ScriptCallback.debug but it is not really defined what it actually should do. If it supposed to just display a shape in transparent red (i.e. `show_object(...,color=(1,0,0,.5))` maybe it does not make sense to define it as a separate function. 

I asked this question because I wanted to make CQ-editor CQGI compliant. I'll redefine it according to the suggestion of @jmwright but indeed let's keep this discussion open.
NB: There is a separate visual debugger with breakpoints/stepping and a CQ stack inspector in the CQ-editor but I would not like to connect those to the `debug()` call.

jmwright, 2020-04-06:
@adam-urbanczyk IIRC, the addition of `debug` to CQGI was just a first pass at making scripts easier to debug. That's basically how I viewed it when I implemented that functionality for the FreeCAD workbench. If the feature needs to evolve, I think that's fine.