CadQuery/cadquery #315: Time for new release?

adam-urbanczyk, 2020-04-08:
@jmwright @dcowden I think it would be good release 2.0 now. I have the feeling that we are keeping on adding functionality without setting a clear release goal. The 2.0RC project was closed long time ago and I recently took the liberty to remove a couple of annoying bugs ( https://github.com/CadQuery/cadquery/projects/5 ). After 2.0 is out I'd propose to focus on moving to OCCT7.4. Let me know what you think!

zignig, 2020-04-08:
2.0 ! should be go with a view to integrating occt ( via pywrap ) for 2.1 

Looking forward to reintegrating cqparts and friends !

zignig, 2020-04-08:
or just go for occt for 2.0, my preferred option... ;)

jmwright, 2020-04-08:
If we're going to move to pywrap/OCP for 2.1, which is probably where we're headed, users will lose access to the Qt and Wx based visualization tools that PythonOCC has built in that OCP won't. I like that OCP doesn't include extra plumbing like that, but we will probably have a few users that are confused. That said, we can't wait forever to do the 2.0 release, and I think that we do have enough other visualization options that we should be able to help users transition to. I think we can navigate the move ok. I'm fine with releasing 2.0 now.

zignig, 2020-04-10:
upon reflection , ( and some issues I am having with OCP ) , I think that a 2.0 release now , willing to help ( I have some free time at the moment ;) )

adam-urbanczyk, 2020-04-10:
@zignig OCP is not ready yet (BTW please open an issue if you found smth) and 2.0 release is long overdue I think.

jmwright, 2020-04-11:
@adam-urbanczyk What steps are required to make this release happen? Will there be any non-source release assets since we're relying on conda?

adam-urbanczyk, 2020-04-11:
The build triggers on master commits and tags. So if you make relase things should work out of the box. You could try making RC2 just to verify if everything still works as expected.

Relevant files:
https://github.com/CadQuery/cadquery/blob/master/azure-pipelines.yml
https://github.com/CadQuery/conda-packages/blob/master/conda-build.yml

jmwright, 2020-04-11:
@adam-urbanczyk I just pushed an 2.0RC2 tag, filled in some release info, and published the release.

https://github.com/CadQuery/cadquery/releases/tag/2.0RC2

jmwright, 2020-04-14:
The publish of the conda package seems to have worked fine: https://anaconda.org/cadquery/cadquery

Azure Pipelines seems to have worked as well.

jmwright, 2020-04-15:
@adam-urbanczyk Are you ready for me to go through and update the changelog, version references, etc and publish the 2.0 release? I'd like to do that before merging #317 in case I introduced a subtle bug that I'm not noticing/anticipating. I'd rather have the whole 2.1 release cycle to work that out rather than putting it into the stable 2.0 release at the last minute.

adam-urbanczyk, 2020-04-15:
Yeah @jmwright  the CI/CD piping seems to work fine. Indeed I'd propose to leave #317 out of that one. What do you think about #321 ? Should it wait till after 2.0?

jmwright, 2020-04-15:
I'll probably regret saying this later, but I think #321 should be fine to merge before the 2.0 release.

jmwright, 2020-04-24:
The CadQuery 2.0 release has been published: https://github.com/CadQuery/cadquery/releases/tag/2.0

Thanks everyone!

closed 2020-04-24 by jmwright