CadQuery/cadquery #318: Best way to build a 3d volume from mixed spline and straight edges

Shimwell, 2020-04-13:
I am trying to creating a 3d shape with edges made from a mixture of splines and straight lines

I have tried this using two methods and have hit dead ends with both ways.

I am able to make the corrected edges, a wire and even a face for the shape but I can't figure out how to extrude the face to create a 3d volume.

```
points = [(900, 300), #0
          (900, -300), #1
          (300, -300), #2
          (200, 0), #3
          (300, 300), #4
          (900, 300)#5 is the same as #0
          ]

e1 =cq.Edge.makeLine(cq.Vector(points[0]),cq.Vector(points[1]))
e2 =cq.Edge.makeLine(cq.Vector(points[1]),cq.Vector(points[2]))
e3 =cq.Edge.makeSpline([cq.Vector(points[2]),cq.Vector(points[3]),cq.Vector(points[4])])
e4 =cq.Edge.makeLine(cq.Vector(points[4]),cq.Vector(points[5]))

edges = [e1,e2,e3,e4]
wire = cq.Wire.assembleEdges(edges)
face = cq.Face.makeFromWires(wire)
show_object(face)
# not sure how to extrude, no face.extrude method
```

Alternatively I have tried to make the 3d shape by copying the [spline](https://cadquery.readthedocs.io/en/latest/examples.html#defining-an-edge-with-a-spline) example, but this method gives the wrong outline of the shape. The spline example has straight line segments in but the spline appears to start in a different point to the end of the lineTo edges which I thought was strange. I have also tried using polyline to combine several lineTo calls to no avail.

```
r = cq.Workplane('XY')
r = r.lineTo(points[0][0], points[0][1])
r = r.lineTo(points[1][0], points[1][1])
r = r.spline(points[2:4])   # I have tried this with and without setting includeCurrent=True
r = r.lineTo(points[5][0], points[5][1])
r = r.close()
r =r.extrude(distance=5)
show_object(r)
```

Any suggestions on the best way to get a mixed spline, currently I can get the right shape but don't know how to extrude or the wrong shape which is easy to extrude.

Many thanks

[image]
[image]

jmwright, 2020-04-13:
@Shimwell The following works for me.

```python
import cadquery as cq

points = [(900, 300), #0
          (900, -300), #1
          (300, -300)
          #(900, 300)#5 is the same as #0
          ]

spline_pts = [ (300, -300), #2
          (200, 0), #3
          (300, 300)]

result = (cq.Workplane("XY")
                    .moveTo(points[0][0], points[0][1])
                    .lineTo(points[1][0], points[1][1])
                    .lineTo(points[2][0], points[2][1])
                    .spline(spline_pts)
                    .close()
                    .extrude(1))

show_object(result)
```

You could probably also work a `polyline` call in there too instead of using `lineTo`, but it depends on what you want.

Shimwell, 2020-04-13:
Thanks so much this works it perfectly, I shall write up the polyline and post if here in case it is useful to anyone else.

It looks like I was missing that moveTo part

Much appreciated @adam-urbanczyk

closed 2020-04-13 by Shimwell