CadQuery/cadquery #322: Exporter producing 0 size stl files from 360 degree rotations of splines

Shimwell, 2020-04-16:
Hi all,

I have been checking out the exporters in a little more detail and came across this possible bug that I found interesting.

When producing a 360 rotation from a spline the resulting stl file is 0 bytes and contains no geometry.

Here is a minimal example

```
import cadquery as cq
from cadquery import exporters

solid = (cq.Workplane("YZ")
          .spline([(100, 100), (100, 50), (50, 50), (50, 100), (100, 100)])
          .close()
          .revolve(360))

with open('rotated_spline.stl', 'w') as f:
    exporters.exportShape(solid, 'STL', f, 0.001)
```

Also this appears to be unique to spline as when I rotate a polyline to 360 degrees then every thing works as expected and a 360 degree shape is produced

ps this a fantastic package, super accessible and so many great unique features

jmwright, 2020-04-16:
This is probably related to or a duplicate of #259

Shimwell, 2020-04-16:
Oh yes, sorry I missed that

Shimwell, 2020-04-16:
Looking forward to OCC 7.4 then. Perhaps best to close this issue as it is covered elsewhere

closed 2020-04-16 by Shimwell

jmwright, 2020-04-16:
Please post your findings to #259 if you have time.