CadQuery/cadquery #324: How to know which version of cadquery I run?

lalebarde, 2020-04-16:
`cadquery --version` is ignored and starts _cadquery_. 

The only solution I found is `conda list | grep cadquery`

This is not a big issue of course, but that would be nice to have the `--version` option available on the command line. It could provide also the [OCC version it relies on](https://github.com/CadQuery/cadquery/issues/274).

jmwright, 2020-04-16:
It doesn't address your desire to see the OCC version, but the following will give you the CadQuery version. Unfortunately, I forgot to update that version number when I tagged RC2. That will be fixed for the 2.0 release in a few days.

```python
>>> import cadquery
>>> print(cadquery.__version__)
'2.0RC1'
```

lalebarde, 2020-04-16:
Thanks Jeremy

closed 2020-04-16 by lalebarde