CadQuery/cadquery #327: How to loft wires defined in independent planes?

lalebarde, 2020-04-19:
Hi, 

[Here](https://github.com/CadQuery/cadquery/issues/319), I could loft from two wires, the first one defined in the workplane, and the second one in a workplane obtained by applying `transformed(rotate=(0, 0, 0), offset=(0, 0, 0))` to the first one. 

But my use case is to loft wires defined in independent planes defined with:

`cq.Workplane(cq.Plane(p[i], n[i], h[i]))`

and not relative one to the other. Here is the simplified pseudo code:

```
import cadquery as cq
wp1 = cq.Workplane("XY").rect(1,2).workplane()
wp2 = cq.Workplane("YZ").rect(1,2).workplane()
wp1.add(wp2) # adds wp2 to the wp1 stack in order to have two wires
result = wp1.loft(combine=True)
```
Result is:

> ValueError: more than one wire is required

In addition, `add` is defined for `class cadquery.CQ(obj)`, not `class cadquery.Workplane(inPlane, origin=(0, 0, 0), obj=None)`. Surprisingly, `wp1.add(wp2)` does not raise an error.

I have considered several solutions:

- Reuse a relative workplane with `transformed(rotate=(0, 0, 0), offset=(0, 0, 0))` like I did in the link on top of this post, but I am not happy with it because my workplanes are defined with absolute origins and normal vectors, built from rotation matrix of the kind of `class cadquery.Matrix(matrix=None)`. This is complex and not wanted.

- Hack the stack like above, but I don't know how. Nevertheless, it is the desired solution.

Is it possible and how please? Alternatively, being able to use `transformed(rotate=(0, 0, 0), offset=(0, 0, 0))` with a `class cadquery.Matrix(matrix=None)` would simplify the first solution.

adam-urbanczyk, 2020-04-19:
Currently `add` does not merge `ctx.pendingWires` which is where wires are taken from for lofting. It could be a bug or a feature (TBD). You can merge wires by simply extending the list:
```python
wp1.ctx.pendingWires.extend(wp2.ctx.pendingWires)
res = wp1.loft()
```

lalebarde, 2020-04-19:
Great, thank you Adam, and I purge the pendingWires after the loft and before the next iteration:
```
        wp1.ctx.pendingWires = []
        wp2.ctx.pendingWires = []
```

closed 2020-04-19 by lalebarde