CadQuery/cadquery #328: Possible bug in the Workplane method transformed(rotate=(0, 0, 0), offset=(0, 0, 0))

lalebarde, 2020-04-19:
This code:

```
import cadquery as cq
from jupyter_cadquery.cadquery import show
triangle = [[-1, 0], [1, 0], [0, 4]]
wp1 = cq.Workplane("XY").polyline(triangle).close().workplane()
wp2 = wp1.transformed(rotate=cq.Vector(30, 0, 0), offset=cq.Vector(0, 0, 0)).polyline(triangle).close().workplane()
result = wp2.loft(combine=True)
show(result)
```
results in:
[image]

There is an unexpected decay of 1 on the Y axis. If I take a triangle 100x bigger, the decay is 100.

Expected result looks like:

[image]

Artificially obtained with `offset=cq.Vector(0, -1.6, 0)`

lalebarde, 2020-04-19:
With `rotate=cq.Vector(0, 0, 0), offset=cq.Vector(0, 0, 3)`, I have also the problem:

[image]

adam-urbanczyk, 2020-04-19:
First call to `workplane` changes the origin (you could use `centerOption='ProjectedOrigin'` or just skip `workplane()`). You probably want this:
```python
import cadquery as cq

triangle = [[-1, 0], [1, 0], [0, 4]]
wp1 = cq.Workplane("XY").polyline(triangle).close()
wp2 = wp1.transformed(rotate=cq.Vector(30, 0, 0), offset=cq.Vector(0, 0, 0)).polyline(triangle).close().workplane()
result = wp2.loft(combine=True)

show_object(result)
```

lalebarde, 2020-04-19:
Thank you Adam, I should have known such a rough bug couldn't be

closed 2020-04-19 by lalebarde