CadQuery/cadquery #334: makeCompound needed to save all objects in STL

bragostin, 2020-04-29:
Why does this yield only the first object in the stl file as if ``each()`` had been ignored
```python
def air_fin(pos):
    return fin.translate(pos).val()

x_fin = cq.Workplane('XY').pushPoints(xy_af).each(air_fin)
x_fin.val().exportStl("test.stl", 0.05)
```
while this saves correctly all the patterned objects in the file?
```python
def air_fin(pos):
    return fin.translate(pos).val()
        
x_fin = cq.Workplane('XY').pushPoints(xy_af).each(air_fin)
x_fin = cq.Compound.makeCompound([o.vals()[0] for o in x_fin.all()])
x_fin.exportStl("test.stl", 0.05)
```

adam-urbanczyk, 2020-04-29:
`val()` return first value from the current `objects`. `cq.exporters.exportShape` does the same actually so I'd say it is a (functional) bug.

adam-urbanczyk, 2020-04-29:
This is related to #212 and #216

bragostin, 2020-04-30:
Then maybe a new ''exportStep'' (or ''exportStl'') could be in this form:
```python
    def exportStep(self, fileName):
        """
        surface_curve_mode:
            0: write without pcurves (2 times smaller STEP file)
            1 (Default): write with pcurves
        """
      
        obj = Compound._makeCompound([o.vals()[0] for o in self.all()])
        c = STEPControl_Controller()
        c.Init() 
        Interface_Static_SetCVal("write.step.schema", "AP214")
        Interface_Static_SetIVal('write.surfacecurve.mode', 0)
        writer = STEPControl_Writer()
        writer.Transfer(obj.wrapped, STEPControl_AsIs)
        
        return writer.Write(fileName)
``` 
This would save all the objects in the stack I presume, and it also gives control on more of the export STEP options of Opencascade. For example setting ''surfacecurve.mode'' to 0 divides the file size by 2.

adam-urbanczyk, 2020-04-30:
I guess yes, the place to put this change would be here:
https://github.com/CadQuery/cadquery/blob/1491e0ccf446a9cbeb4a6d7b19f0922918ba648d/cadquery/occ_impl/exporters.py#L70
We need to rethink the exporters API anyhow, the possibility to control `surfacecurve.mode` looks like a good idea.

michaelgale, 2020-05-25:
I would definitely welcome the ability to pass parameters to the exporters.  I've implemented my own STEP, IGES, and STL exporters because of the requirement to change items such as surface_curve_mode, tolerances, etc.  Especially for file formats which require pre-meshing--the ability to control the resolution and meshing strategy is essential.  

Lastly, my STEP exporter injects more meta data into the header such as author, description, copyright, contact info, etc. This is both nice-to-have and helpful when sharing STEP files.  Furthermore, I re-render all floating point ASCII values in the STEP file using the `Decimal` class so that nuisance precision values such as "1E-37" or "8.9999999995" don't inflate the STEP file with needless bytes.  For many 10's MB files, this really adds up!  It also "cleans" up the STEP file so that importers will have a better time making internal representations of the geometry--presumably by ensuring surface normals snap identically with a desired axis for example, rather than invisibly (but numerically) slightly off-axis.

jmwright, 2020-05-25:
@michaelgale Thanks for sharing. Is your exporter code available anywhere so that you can add links to it here? Those all sound like useful things to have in the CadQuery exporters.

michaelgale, 2020-05-25:
@jmwright Currently that code is bundled in a proprietary (and messy!) API, but I do intend extract all the useful general purpose code into its own public repo.  This includes not just file I/O, but I've made many new Selector classes, shape builders, etc. which hopefully the community would find useful.

The dilemma is whether these capabilities get bolted into CQ (cq.py is getting **very** long) or a standalone utility toolkit designed to extend CQ?  In any case, I would be happy to contribute.  I'm not sure what the roadmap for the CQ "core" is, but perhaps a README or project could simply itemize the plan--even a bullet point list of desired new features, perhaps in the form of method signatures (with no code!)

adam-urbanczyk, 2020-05-25:
Just share what you have and let's take it from there @michaelgale . I personally don't worry that much about LOCs.

adam-urbanczyk, 2020-08-02:
Solved by #415:
```python
import cadquery as cq

res = (
    cq.Workplane()
    .rect(2,2,forConstruction=True)
    .vertices().circle(.5).extrude(1)
)

cq.exporters.export(res,'test.step')
cq.exporters.export(res,'test.step')
```

closed 2020-08-02 by adam-urbanczyk