CadQuery/cadquery #336: Very bad performance on simple mirror union

goatchurchprime, 2020-04-30:
I hit this problem making a wing model by doing a loft between aerofoil sections for one wing and then reflecting about the main axis and applying union.  

The problem here is that the `union()` takes far to long to execute as we add more wing-sections, even though the sections are nowhere near the intersection.  

I've used a faceted oval in this example, because we don't need an aerofoil.  

```Python
import cadquery as cq
import math

Nsegmentsaround, Nsegmentsalong = 50, 4

def cycle(r, rad):
    r = r.moveTo(0, 0)
    for i in range(1, Nsegmentsaround):
        th = math.radians(i/Nsegmentsaround*360)
        x, y = rad*(1-math.cos(th)), 0.2*math.sin(th)*rad
        r = r.lineTo(x, y)
    r = r.close()
    return r

r = cq.Workplane("XZ", (0,0,0))
r = cycle(r, 0.6)
for j in range(Nsegmentsalong):
    r = r.workplane(offset=0.2, centerOption="ProjectedOrigin")
    r = cycle(r, 1/(j+2))
r = r.loft(combine=True)

m = r.mirror(mirrorPlane='XZ', basePointVector=(0,0,0))
r = m.union(r)  # <-- this takes too much time as Nsegmentsalong increases
```

* Nsegmentsalong=2 takes 18 seconds
* Nsegmentsalong=3 takes 47 seconds
* Nsegmentsalong=4 takes 96 seconds
* Nsegmentsalong=5 takes 168 seconds

Basically, it's taking almost double the time as I add each section on the end nowhere near the unioned face.  This is why the model cannot complete with a design that has 10 sections and 90 points around the perimeter.  

[image]

(Side note: when recreating this issue I found a [bug in jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery/issues/14))

adam-urbanczyk, 2020-05-01:
I cannot reproduce on the OCP branch for N=5 (see below). Could you `%prun` the offending code and share the result? BTW: you might want to use ellipse primitive instead of approximating with polylines yourself. For shapes that are not implemented it is better to use `parametricCurve`.

```python
%time res = m.union(r)
CPU times: user 11.2 s, sys: 416 ms, total: 11.6 s
Wall time: 1.75 s
```

```python
%prun res = m.union(r)

         2401 function calls in 1.726 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    1.696    1.696    1.696    1.696 shapes.py:582(_bool_op)
        1    0.010    0.010    0.010    0.010 shapes.py:223(clean)
        1    0.005    0.005    1.726    1.726 cq.py:2717(union)
      192    0.005    0.000    0.005    0.000 geom.py:26(__init__)
...
```

goatchurchprime, 2020-05-01:
The 2^N speed problem reproduces in the [launch-binder](https://mybinder.org/v2/gh/bernhard-42/jupyter-cadquery/master?urlpath=lab&filepath=examples%2Fcadquery.ipynb) demo page that links from the [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery) repo.  (See image below of it.)

I was actually using the shape of an aerofoil, which is defined as a contour of 150 points from a book, but I found I could reproduce the issue using the polyline of an ellipse, which makes it a little easier to report the issue.  

%prun ???!!  You learn something new every day!  

Here's what it gives -- it seems to actually be calling the OCC-functions.

```
          3781 function calls in 124.642 seconds
 
    Ordered by: internal time
 
    ncalls  tottime  percall  cumtime  percall filename:lineno(function)
         1   64.055   64.055   64.055   64.055 {built-in method OCC.Core._BRepAlgoAPI.new_BRepAlgoAPI_Fuse}
         1   60.434   60.434  124.489  124.489 shapes.py:596(fuse)
         1    0.140    0.140    0.141    0.141 shapes.py:236(clean)
         1    0.009    0.009  124.642  124.642 cq.py:2656(union)
       192    0.001    0.000    0.001    0.000 geom.py:36(__init__)
        24    0.000    0.000    0.001    0.000 geom.py:740(_calcTransforms)
       192    0.000    0.000    0.000    0.000 {built-in method OCC.Core._gp.gp_Vec_swiginit}
```

Reproducing bugs in CAD where you normally have to operate it through a GUI can be a big challenge, so if this method of doing it with a short script on a launch-binder thing works, it would be a great thing.  

[image]

adam-urbanczyk, 2020-05-01:
OK @goatchurchprime , so most of the time is spend in fuse. Are you able install CQ locally (via conda) and test? I don't know which version is on binder and there were changes to the bool ops in the recent past. If that does not help you'll  need to use the OCP branch.

Regarding airfoils - you might be interested in using a spline instead of a polyline.

goatchurchprime, 2020-05-01:
I originally identified the problem on my own install of CQ which reports `cq.__version__ = '2.0.0dev'`

The launch.binder code page reports it as `cq.__version__ = '2.0RC1'`

I can't explain why 2.0.0dev is missing from the list of tags on the repo, and I don't know which is the more recent.  

If it worked, the launch.binder thing would be a very useful resource for users to narrow down bugs (especially if you could load it with different versions), when not all of us are in the best position to know how to build and run these complex systems from source.  

The aerofoils come from a [historical database at the University of Illinois](https://m-selig.ae.illinois.edu/ads.html) which gives them as sequences of points for the upper and lower surfaces.  I intend to thin them to use a small number of splines, but haven't found any code that does this sort of thing yet (and would only write it myself as a last resort when I am absolutely sure it's not available already, which can't be possible for something so useful).

This is not a blocker for me, but I am reporting it because it seems to show up a problem if an algorithm is doubling in time without a reasonable explanation.  I've written a lot of CAM software in the past, so I know how these things can arise in CAD kernels.

adam-urbanczyk, 2020-05-02:
I can confirm that on hte current master (thus with OCCT 6.9) this code is rather slow. It gets much faster with OCCT 7.4 (so with the OCP branch).

You can make it even faster (even on 6.9) if you'll use splines:
```python
import cadquery as cq
import math

Nsegmentsaround, Nsegmentsalong = 150, 15

def cycle(r, rad):
    
    pts = [(0,0)]
    
    for i in range(1, Nsegmentsaround):
        th = math.radians(i/Nsegmentsaround*360)
        x, y = rad*(1-math.cos(th)), 0.2*math.sin(th)*rad
        pts.append((x,y))
    
    r = r.spline(pts,
                 periodic=True,
                 makeWire=True)
    
    return r

r = cq.Workplane("XZ", (0,0,0))
r = cycle(r, 0.6)

for j in range(Nsegmentsalong):
    r = r.workplane(offset=0.2, centerOption="ProjectedOrigin")
    r = cycle(r, 1/(j+2))

r = r.loft(combine=True)

m = r.mirror(mirrorPlane='XZ', basePointVector=(0,0,0))
r = m.union(r)
```
[image]

This code runs in a couple of seconds:
```
%time r = m.union(r)
CPU times: user 4.48 s, sys: 1.1 s, total: 5.58 s
Wall time: 2.84 s
```

adam-urbanczyk, 2020-06-04:
@goatchurchprime did you try on the current master? For 10 segments I get:
```python
%time fop.Build()
CPU times: user 9.13 s, sys: 46.5 ms, total: 9.18 s
Wall time: 8.26 s
```
What is even more interesting, in the newer OCCT there is a glue option in the bool ops which is meant to handle such cases:
```python
fop.SetGlue(OCP.BOPAlgo.BOPAlgo_GlueEnum.BOPAlgo_GlueFull)
%time fop.Build()
CPU times: user 192 ms, sys: 0 ns, total: 192 ms
Wall time: 191 ms
```

goatchurchprime, 2020-06-17:
I haven't had time and I already had a work-around.  I used to write CAD software, so I know that easily reproduced problems like this are essential for debugging.  Adding geometry that is very far from the mating surface cannot lose performance on this scale unless there is something badly wrong.  Finding a way round it leaves the problem in place, and it will reappear in more complex parts where it will be much harder to put your finger on the issue.

adam-urbanczyk, 2020-06-17:
I'm not sure what's the message you are trying to convey @goatchurchprime . I already mentioned multiple times that I cannot reproduce your scaling numbers on master/OCCT 7.4.  Additionally, current fastest method for achieving what you want is using the glue mode. If you have ideas how to make the CAD kernel even faster, feel free to open issues or submit patches here: https://dev.opencascade.org/ .

closed 2020-06-17 by adam-urbanczyk

bernhard-42, 2020-08-16:
@adam-urbanczyk just for info, since the question came up: 
The latest version for Jupyter Cadquery uses CadQuery 2.0.0 as installed by `conda install -c cadquery cadquery=2.0.0` (see https://github.com/bernhard-42/jupyter-cadquery/blob/master/environment.yml).
I had used cadquery's `master` before, but am now on the official 2.0 release and will keep the binder and docker container of jupyter-cadquery on the latest official release of your cadquery.

adam-urbanczyk, 2020-08-17:
Thanks for the clarification @bernhard-42