CadQuery/cadquery #337: Cannot find correct font resource in makeText (OCP branch)

michaelgale, 2020-05-01:
When attempting to specify a specific typeface for makeText, it results in the following error:

```shell
Font_FontMgr, warning: unable to find font '/usr/share/fonts/truetype/DIN/DIN-Regular.ttf' [regular]; 'FreeSans' [aspects: regular,bold,italic,bold-italic] [paths: /usr/share/fonts/truetype/freefont/FreeSans.ttf;/usr/share/fonts/truetype/freefont/FreeSansBold.ttf;/usr/share/fonts/truetype/freefont/FreeSansOblique.ttf;/usr/share/fonts/truetype/freefont/FreeSansBoldOblique.ttf] is used instead
```

After making an alternate implementation of makeText, I discovered that the `makeText` method should be modified as follows:

```python
        text_flat = Shape(
            builder.Perform(font.FontPath(font_kind).ToCString(), size, font_kind, text)
        )
```

should be:

```python
        text_flat = Shape(
            builder.Perform(font.FontName().ToCString(), size, font_kind, text)
        )
```

Perhaps its related to how `Font_BRepTextBuilder` gets wrapped when building OCP?  

Tested on Linux version 4.15.0-96-generic (ubuntu 18.04)

jmwright, 2020-05-01:
I get the same error on Ubuntu 18.04.

michaelgale, 2020-05-01:
@jmwright Glad to hear its repeatable. After changing the way `Font_BRepTextBuilder` gets called, I can use any font (although not all Font Aspects) discovered by OCCT from the OS's font libraries.  I can also register new fonts (or modify mappings inside the OCCT internal font library) to fine tune the selection of desired typefaces.  I think the problem is related to how the arguments to `Font_BRepTextBuilder.Perform` get mapped from OCCT to OCP.

adam-urbanczyk, 2020-05-02:
Thanks for finding this @michaelgale !

jmwright, 2021-02-03:
So was this fixed in this commit? https://github.com/CadQuery/cadquery/commit/23418b98a7278173eb7576ed5e3faf9ec5a803ea

jmwright, 2024-04-29:
This seems to be fixed. Closing.

closed 2024-04-29 by jmwright