CadQuery/cadquery #338: Which options for show_object?

lalebarde, 2020-05-03:
Hi, 
` show_object` enables to set colors on objects, but I don't find information on other options. Where can I find the full list please? For example, I would like to use dotted or dashed lines, or different line thickness - for debugging, I project 2D drawings on a wall.

jmwright, 2020-05-04:
`show_object` is implemented by the execution environment (CQ-editor, jupyter-cadquery, etc), and so the options supported can vary. For CQ-editor, I believe only color and alpha options are available.

Reference: https://github.com/CadQuery/CQ-editor/pull/127/files

zwn, 2024-02-16:
It would be great [typing stubs](https://peps.python.org/pep-0484/#stub-files) for `show_object` could be provided to help linters and editors.