CadQuery/cadquery #344: ModuleNotFoundError: No module named 'OCC.Core'

roipoussiere, 2020-05-06:
In order to produce a flatpak, I'm trying to install cadquery from sources. This is a long work (it require to install oce, python-occ and other dependencies from sources) but I think I can see the light at the end of the tunnel. :)

I get the same error as #150 when I try to import cadquery, but I prefer to open a new issue as the other one is fixed.

```python
>>> import OCC
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/app/lib/python3.7/site-packages/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix
  File "/app/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 3, in <module>
    from OCC.Core.gp import (
ModuleNotFoundError: No module named 'OCC.Core'
```

Installed dependencies:

- glu 9.0.1
- freeglut 3.2.1
- oce 0.18.3
- swig 3.0.12
- pythonocc-core 0.18.1 (cadquery fork)
- pyparsing 2.4.7
- cadquery 2.0

<details>
<summary>ls /app/lib/python3.7/site-packages/OCC/</summary>
<pre>
Adaptor2d.py	   _BRepCheck.so	    _Dico.so		GeomConvert.py	     _Interface.so	Precision.py	     _StdFail.so	 _TDF.so
_Adaptor2d.so	   BRepClass3d.py	    Display		_GeomConvert.so       Intf.py		_Precision.so	     StdPrs.py		 TDocStd.py
Adaptor3d.py	   _BRepClass3d.so	    Draft.py		GeomFill.py	     _Intf.so		Primitives.py	     _StdPrs.so	 _TDocStd.so
_Adaptor3d.so	   BRepClass.py	    _Draft.so		_GeomFill.so	     IntImpParGen.py	_Primitives.so       StdSelect.py	 TFunction.py
Addons.py	   _BRepClass.so	    DsgPrs.py		GeomInt.py	     _IntImpParGen.so	ProjLib.py	     _StdSelect.so	 _TFunction.so
_Addons.so	   BRepExtrema.py	    _DsgPrs.so		_GeomInt.so	     IntImp.py		_ProjLib.so	     StepAP203.py	 TNaming.py
AdvApp2Var.py	   _BRepExtrema.so	    Dynamic.py		GeomLib.py	     _IntImp.so	Prs3d.py	     _StepAP203.so	 _TNaming.so
_AdvApp2Var.so	   BRepFeat.py		    _Dynamic.so	_GeomLib.so	     IntPatch.py	_Prs3d.so	     StepAP209.py	 TopAbs.py
AdvApprox.py	   _BRepFeat.so	    ElCLib.py		GeomLProp.py	     _IntPatch.so	PrsMgr.py	     _StepAP209.so	 _TopAbs.so
_AdvApprox.so	   BRepFilletAPI.py	    _ElCLib.so		_GeomLProp.so	     IntPolyh.py	_PrsMgr.so	     StepAP214.py	 TopBas.py
AIS.py		   _BRepFilletAPI.so	    ElSLib.py		GeomPlate.py	     _IntPolyh.so	__pycache__	     _StepAP214.so	 _TopBas.so
_AIS.so		   BRepFill.py		    _ElSLib.so		_GeomPlate.so	     IntPoly.py	Quantity.py	     StepBasic.py	 TopClass.py
AppBlend.py	   _BRepFill.so	    ExprIntrp.py	GeomProjLib.py	     _IntPoly.so	_Quantity.so	     _StepBasic.so	 _TopClass.so
_AppBlend.so	   BRepGProp.py	    _ExprIntrp.so	_GeomProjLib.so       IntRes2d.py	Resource.py	     STEPCAFControl.py   TopCnx.py
AppCont.py	   _BRepGProp.so	    Expr.py		Geom.py		     _IntRes2d.so	_Resource.so	     _STEPCAFControl.so  _TopCnx.so
_AppCont.so	   BRepIntCurveSurface.py   _Expr.so		_Geom.so	     Intrv.py		RWStepAP203.py       STEPConstruct.py	 TopExp.py
AppDef.py	   _BRepIntCurveSurface.so  Extrema.py		GeomTools.py	     _Intrv.so		_RWStepAP203.so      _STEPConstruct.so   _TopExp.so
_AppDef.so	   BRepLib.py		    _Extrema.so	_GeomTools.so	     IntStart.py	RWStepAP214.py       STEPControl.py	 TopLoc.py
AppParCurves.py     _BRepLib.so		    FairCurve.py	GeomToStep.py	     _IntStart.so	_RWStepAP214.so      _STEPControl.so	 _TopLoc.so
_AppParCurves.so    BRepLProp.py	    _FairCurve.so	_GeomToStep.so	     IntSurf.py	RWStepBasic.py       StepDimTol.py	 TopoDS.py
ApproxInt.py	   _BRepLProp.so	    FEmTool.py		gp.py		     _IntSurf.so	_RWStepBasic.so      _StepDimTol.so	 _TopoDS.so
_ApproxInt.so	   BRepMAT2d.py	    _FEmTool.so	GProp.py	     IntTools.py	RWStepDimTol.py      STEPEdit.py	 TopoDSToStep.py
Approx.py	   _BRepMAT2d.so	    FilletSurf.py	_GProp.so	     _IntTools.so	_RWStepDimTol.so     _STEPEdit.so	 _TopoDSToStep.so
_Approx.so	   BRepMesh.py		    _FilletSurf.so	_gp.so		     IntWalk.py	RWStepElement.py     StepElement.py	 TopOpeBRepBuild.py
AppStdL.py	   _BRepMesh.so	    FSD.py		GraphDS.py	     _IntWalk.so	_RWStepElement.so    _StepElement.so	 _TopOpeBRepBuild.so
_AppStdL.so	   BRepOffsetAPI.py	    _FSD.so		_GraphDS.so	     Law.py		RWStepFEA.py	     StepFEA.py	 TopOpeBRepDS.py
AppStd.py	   _BRepOffsetAPI.so	    GccAna.py		Graphic3d.py	     _Law.so		_RWStepFEA.so	     _StepFEA.so	 _TopOpeBRepDS.so
_AppStd.so	   BRepOffset.py	    _GccAna.so		_Graphic3d.so	     LICENSE		RWStepGeom.py	     StepGeom.py	 TopOpeBRep.py
Aspect.py	   _BRepOffset.so	    GccEnt.py		GraphTools.py	     LocalAnalysis.py	_RWStepGeom.so       _StepGeom.so	 _TopOpeBRep.so
_Aspect.so	   BRepPrimAPI.py	    _GccEnt.so		_GraphTools.so	     _LocalAnalysis.so  RWStepRepr.py	     StepRepr.py	 TopOpeBRepTool.py
AUTHORS		   _BRepPrimAPI.so	    GccGeo.py		HatchGen.py	     LocOpe.py		_RWStepRepr.so       _StepRepr.so	 _TopOpeBRepTool.so
Bisector.py	   BRepPrim.py		    _GccGeo.so		_HatchGen.so	     _LocOpe.so	RWStepShape.py       STEPSelections.py   TopTools.py
_Bisector.so	   _BRepPrim.so	    GccInt.py		Hatch.py	     LProp3d.py	_RWStepShape.so      _STEPSelections.so  _TopTools.so
BiTgte.py	   BRepProj.py		    _GccInt.so		_Hatch.so	     _LProp3d.so	RWStepVisual.py      StepShape.py	 TopTrans.py
_BiTgte.so	   _BRepProj.so	    GccIter.py		Hermit.py	     LProp.py		_RWStepVisual.so     _StepShape.so	 _TopTrans.so
BlendFunc.py	   BRep.py		    _GccIter.so	_Hermit.so	     _LProp.so		RWStl.py	     StepToGeom.py	 TPrsStd.py
_BlendFunc.so	   _BRep.so		    GCE2d.py		HLRAlgo.py	     MAT2d.py		_RWStl.so	     _StepToGeom.so	 _TPrsStd.so
Blend.py	   BRepSweep.py	    _GCE2d.so		_HLRAlgo.so	     _MAT2d.so		Select3D.py	     StepToTopoDS.py	 TShort.py
_Blend.so	   _BRepSweep.so	    gce.py		HLRAppli.py	     Materials.py	_Select3D.so	     _StepToTopoDS.so	 _TShort.so
BndLib.py	   BRepTools.py	    _gce.so		_HLRAppli.so	     _Materials.so	SelectBasics.py      StepVisual.py	 UnitsAPI.py
_BndLib.so	   _BRepTools.so	    GCPnts.py		HLRBRep.py	     math.py		_SelectBasics.so     _StepVisual.so	 _UnitsAPI.so
[... 10 lines cut ...]
BOPTools.py	   ChFi2d.py		    _Geom2dGcc.so	_Image.so	     _NCollection.so	ShapeCustom.py       TColGeom2d.py	 Voxel.py
_BOPTools.so	   _ChFi2d.so		    Geom2dHatch.py	IncludeLibrary.py     NIS.py		_ShapeCustom.so      _TColGeom2d.so	 _Voxel.so
BRepAdaptor.py	   ChFi3d.py		    _Geom2dHatch.so	_IncludeLibrary.so    _NIS.so		ShapeExtend.py       TColGeom.py	 XBRepMesh.py
_BRepAdaptor.so     _ChFi3d.so		    Geom2dInt.py	__init__.py	     NLPlate.py	_ShapeExtend.so      _TColGeom.so	 _XBRepMesh.so
BRepAlgoAPI.py	   ChFiDS.py		    _Geom2dInt.so	IntAna2d.py	     _NLPlate.so	ShapeFix.py	     TColgp.py		 XCAFApp.py
_BRepAlgoAPI.so     _ChFiDS.so		    Geom2dLProp.py	_IntAna2d.so	     OSD.py		_ShapeFix.so	     _TColgp.so	 _XCAFApp.so
BRepAlgo.py	   ChFiKPart.py	    _Geom2dLProp.so	IntAna.py	     _OSD.so		ShapeProcessAPI.py   TCollection.py	 XCAFDoc.py
_BRepAlgo.so	   _ChFiKPart.so	    Geom2d.py		_IntAna.so	     PCDM.py		_ShapeProcessAPI.so  _TCollection.so	 _XCAFDoc.so
BRepApprox.py	   Contap.py		    _Geom2d.so		IntCurve.py	     _PCDM.so		ShapeProcess.py      TColQuantity.py	 XCAFPrs.py
_BRepApprox.so	   _Contap.so		    GeomAbs.py		IntCurvesFace.py      Plate.py		_ShapeProcess.so     _TColQuantity.so	 _XCAFPrs.so
BRepBlend.py	   Convert.py		    _GeomAbs.so	_IntCurvesFace.so     _Plate.so		ShapeUpgrade.py      TColStd.py	 XSControl.py
_BRepBlend.so	   _Convert.so		    GeomAdaptor.py	_IntCurve.so	     PLib.py		_ShapeUpgrade.so     _TColStd.so	 _XSControl.so
BRepBndLib.py	   CPnts.py		    _GeomAdaptor.so	IntCurveSurface.py    _PLib.so		SortTools.py	     TDataStd.py
_BRepBndLib.so	   _CPnts.so		    GEOMAlgo.py	_IntCurveSurface.so   Plugin.py		_SortTools.so	     _TDataStd.so
BRepBuilderAPI.py   CSLib.py		    _GEOMAlgo.so	InterfaceGraphic.py   _Plugin.so	Standard.py	     TDataXtd.py
_BRepBuilderAPI.so  _CSLib.so		    GeomAPI.py		_InterfaceGraphic.so  Poly.py		_Standard.so	     _TDataXtd.so
BRepCheck.py	   Dico.py		    _GeomAPI.so	Interface.py	     _Poly.so		StdFail.py	     TDF.py
</pre>
</details>

If I understand well, `OCC` module has been renamed to `OCC.Core` in the last release ([related commit](https://github.com/CadQuery/cadquery/commit/42f7b691497638b2a3178feeb36d341d87709622)).

So why I still don't have OCC.Core in my site-packages?

Should I need to switch to occ/python-occ 7.4?

roipoussiere, 2020-05-06:
> Should I use the latest git commit instead of using the latest release?

This seems better, except one failure in unit tests (#345).

adam-urbanczyk, 2020-05-06:
No OCC.Core indicates outdated PythonOCC version.

roipoussiere, 2020-05-06:
> No OCC.Core indicates outdated PythonOCC version.

I used [the latest release of the CadQuery fork of pythonocc-core](https://github.com/CadQuery/pythonocc-core/releases/tag/0.18.1).

roipoussiere, 2020-05-07:
What changes are applied to the CadQuery fork of pythonocc-core?

Can I use pythonocc-core directly in order to use the more recent [0.18.2](https://github.com/tpaviot/pythonocc-core/releases/tag/0.18.2)?

The fact is I'm producing a CADquery flatpak and I would like to base it from a release, not from the latest git commit, in order to be sure that the build is reproducible and the app is stable.

adam-urbanczyk, 2020-05-07:
Just use the HEAD of the CQ fork. Changes are related to visualization, If you are only interested in CQ (thus not CQ-editor) then upstream is fine.

roipoussiere, 2020-05-07:
Ok thanks.

closed 2020-05-07 by roipoussiere