CadQuery/cadquery #345: testText fails

roipoussiere, 2020-05-06:
Using the latest code from the git repo (master) in cadquery (a140d1035c5ef058a5c89554ef7390873e10c370) and the python-occ's cadquery fork ([`701e924`](https://github.com/CadQuery/pythonocc-core/commit/701e924ae40701cbe6f9992bcbdc2ef22aa9b5ab)), I ran `python3 -m unittest` and all tests passes except [testText](https://github.com/CadQuery/cadquery/blob/master/tests/test_cadquery.py#L2657):

```
ERROR: testText (test_cadquery.TestCadQuery)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/run/build/cadquery/cadquery/tests/test_cadquery.py", line 2671, in testText
    font="Sans",
  File "/app/lib/python3.7/site-packages/cadquery/cq.py", line 3410, in text
    position=self.plane,
  File "/app/lib/python3.7/site-packages/cadquery/occ_impl/shapes.py", line 1952, in makeText
    bb = text_flat.BoundingBox()
  File "/app/lib/python3.7/site-packages/cadquery/occ_impl/shapes.py", line 352, in BoundingBox
    return BoundBox._fromTopoDS(self.wrapped, tol=tolerance)
  File "/app/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 868, in _fromTopoDS
    return cls(bbox)
  File "/app/lib/python3.7/site-packages/cadquery/occ_impl/geom.py", line 780, in __init__
    XMin, YMin, ZMin, XMax, YMax, ZMax = bb.Get()
RuntimeError: Standard_ConstructionError
Bnd_Box is void
```

adam-urbanczyk, 2020-05-06:
This is not reproducible in our CI (both travis and appveyor are green). I will not be able to debug this on my own. Could you try to check what is in e.g. `text_flat`? The bounding box seems to choke on that according to the pasted traceback.

roipoussiere, 2020-05-06:
One of suspicious line is [`text_flat = Shape(text_to_brep(text, font, font_kind, size, False))`](https://github.com/CadQuery/cadquery/blob/master/cadquery/occ_impl/shapes.py#L195)

So from python-occ, I tried this:

```python
>>> from OCC.Core.Addons import text_to_brep, Font_FA_Regular
>>> text_to_brep('foo', 'Sans', Font_FA_Regular, 0.5, False)
Font Sans initialization failed.
<Swig Object of type 'TopoDS_Shape *' at 0x7fec70863db0>
>>> text_to_brep('foo', 'Arial', Font_FA_Regular, 0.5, False)
Font Arial initialization failed.
swig/python detected a memory leak of type 'TopoDS_Shape *', no destructor found.
<Swig Object of type 'TopoDS_Shape *' at 0x7fec704ec900>
```

So I suppose it comes from this *Font Sans initialization failed.* Maybe FreeType or similar is required?

adam-urbanczyk, 2020-05-10:
It would be interesting to know if the result of `text_to_brep` contains anything at all. Regarding fonts probably OCC does something weird when looking them up - you could try using `strace` to get to the bottom of things.