CadQuery/cadquery #348: Expose 2D fillet and 2D chamfer

adam-urbanczyk, 2020-05-07:


greyltc, 2020-09-07:
I work around this today by `.extrude(1)`ing stuff, doing the operation in 3D and then taking a face off the 3D thing I made.

Irev-Dev, 2020-12-24:
Hey I'm just cross-posting from #551 

Some context from other code-cad packages might help decide on an api.

There I said:

I've been playing around with a [couple of different code-CAD packages](https://github.com/Irev-Dev/curated-code-cad) lately. and CadQuery's `selectors` feature certainly is a unique and valuable feature, But I also really like that CascadeStudio taps into a "Sketch" functionality and I'm able to fillet points as I define them. [here's an example](https://github.com/Irev-Dev/curated-code-cad/blob/main/birdhouse/CascadeStudio.js#L27) making the same bird house as the PR @jmwright mentioned. Since both CascadeStudio and CadQuery use OpenCascade under the hood maybe it could be replicated and maybe worth adding?

Side note, there's similar [functionality in OpenSCAD](https://github.com/Irev-Dev/curated-code-cad/blob/main/birdhouse/OpenSCAD.scad#L13) with the right libraries (uses a `[x, y, radius]` API).

adam-urbanczyk, 2020-12-24:
Thanks @Irev-Dev , but I'm not sure what you are trying to suggest. #42 covers sketching so it is on the roadmap.

Irev-Dev, 2020-12-24:
@adam-urbanczyk 2d constraint sketches would be awesome, and if fillets fits into that, that makes sense.

The suggestion is to add `lineTo().fillet()` type API as that might be a low hanging fruit, i.e. easier to implement than full-blown constrained sketches.

Is your roadmap published somewhere?

jmwright, 2020-12-24:
@Irev-Dev Roadmap is here: https://cadquery.readthedocs.io/en/latest/roadmap.html

PR #683 (2021-03-11): 2D fillet chamfer
Will resolve #348
closed 2021-03-11 by marcus7070